linux-sunxi/sunxi-tools #37: Use sunxi-fel to boot multiple devices in parallel

ThomasKaiser, 2016-03-15:
Maybe I simply miss something (like the sunxi-fel's progress bar I've not noticed for a long time). We changed our build process to be able to do FEL/NFS boot and I wanted now attach up to 4 sunxi boards to my new build host to be able to test them in parallel.

I attached two and get:

    root@armbian:~# sunxi-fel -v ver
    AWUSBFEX soc=00001680(H3) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000

    root@armbian:~# lsusb | grep Onda
    Bus 008 Device 002: ID 1f3a:efe8 Onda (unverified) V972 tablet in flashing mode
    Bus 003 Device 002: ID 1f3a:efe8 Onda (unverified) V972 tablet in flashing mode

Is there already the possibility to specify bus/device ID and is sunxi-fel prepared to be run in multiple instances at the same time?

ssvb, 2016-03-15:
My plan was to [eventually implement a FEL daemon](https://www.mail-archive.com/linux-sunxi@googlegroups.com/msg13468.html), which would do the following:

    we may consider adding support for handling multiple
    connected devices at once. Of course not in the "fel" command
    line tool. But as some sort of a daemon, which waits for connected
    Allwinner devices in FEL mode, retrieves the unique SID value
    from each device and uploads appropriate bootloader images to
    them.

And at least for retrieving the device-unique SID value, this partially depends on the implementation of the readl/writel functionality, which first needs to be reviewed and pushed to git: https://groups.google.com/forum/#!msg/linux-sunxi/FG40i-5d0NY/2pl_M1GIAgAJ

ssvb, 2016-03-15:
Well, specifying the bus/device ID for the sunxi-fel would be also useful. Just the SID seems to be a bit better and more convenient thing for identifying devices.

ThomasKaiser, 2016-03-15:
I agree on using the SID would be better. So based on the (missing) feedback on the ML I would assume you need some feedback regarding the _readl_ call with your patch applied?

Regarding a daemon I'm not sure whether I would prefer being able to use sunxi-fel instead since one my main usecases of this 'FEL boot in parallel' approach is to use different configurations sent to devices of the same type in an automated fashion. Using a daemon I would've to always update the config and sending SIGHUP signals when I want to be able to send different stuff to different devices or how do you try to implement this?

ThomasKaiser, 2016-03-15:
Reading the SID after applying the patch works nicely (tried with pcDuino Nano 3, Banana Pi, Banana Pro, OPi PC and Banana Pi M3 -- the latter storing the SID obviously somewhere else):

    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel ver
    AWUSBFEX soc=00001651(A20) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel readl 0x01c23800
    0x16516608
    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel ver
    AWUSBFEX soc=00001651(A20) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel readl 0x01c23800
    0x1651668d
    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel ver
    AWUSBFEX soc=00001651(A20) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel readl 0x01c23800
    0x1651660d
    
    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel ver
    AWUSBFEX soc=00001673(A83T) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel readl 0x01c23800
    0x00000000
    
    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel ver
    AWUSBFEX soc=00001680(H3) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel readl 0x01c23800
    0x00000000

ssvb, 2016-03-15:
> Using a daemon I would've to always update the config and sending SIGHUP signals when I want to be able to send different stuff to different devices or how do you try to implement this?

Either this, or just replace the boot.scr file and/or the U-Boot and Kernel images at the appropriate locations on your file system. Copying/overwriting the images at the bootable locations can be a part of the U-Boot or Kernel build scripts. Not very much different from what you would do with PXE or BOOTP.

n1tehawk, 2016-03-15:
NextThing had started multiple device support in their fork (e.g. https://github.com/NextThingCo/sunxi-tools/commit/16386a71904baaba3297c6ea97d222824af3ca41), but didn't request it upstream - I also don't recall any related discussions on the mailing list.

ssvb, 2016-03-15:
> Reading the SID after applying the patch works nicely (tried with pcDuino Nano 3, Banana Pi, Banana Pro, OPi PC and Banana Pi M3 -- the latter storing the SID obviously somewhere else).

The whole SID is a 128-bit value, you can check the https://linux-sunxi.org/SID_Register_Guide wiki page. The address is SoC dependent (at least on H3 it is different). There is one exception though: on A31/A31s it is not easily retrievable because Allwinner decided to move the SID from the SoC into the PMIC on these chips.

ssvb, 2016-03-15:
@n1tehawk That's a good find, probably @ThomasKaiser can use their patch until we implement something better.

ssvb, 2016-03-15:
BTW, the U-Boot bootloder derives the Ethernet MAC address from SID. If we use the same algorithm for SID transformation, then we can also use MAC addresses in the FEL daemon config file as an alternative to raw SID values.

ThomasKaiser, 2016-03-15:
@n1tehawk thx that looks good. I will use that for now and get back to you here in this issue in a week or two with possible use cases for multiple device support and experiences made. One thing we want to try out (automatically) is sending the wrong stuff to the device(s) and that might be more simply when being able to use a CLI instead of daemon.

Being able to specify the bus ID I'm currently also able to rely on the SID by probing all buses for devices (if/when I've understood how which address to query on H3 ;) )

ssvb, 2016-03-15:
> Being able to specify the bus ID I'm currently also able to rely on the SID by probing all buses for devices (if/when I've understood how which address to query on H3 ;) )

We can also implement a "sunxi-fel sid" command and take care of the SoC dependent address differences :)

ThomasKaiser, 2016-03-15:
> We can also implement a "sunxi-fel sid" command and take care of the SoC dependent address differences

That would be a useful addition, yes. And BTW I don't get it how to use NextThing's -d/--dev switch correctly (one OPi PC and one Banana Pi connected but only the latter shown):

    root@armbian:/usr/local/src/sunxi-tools# lsusb | grep Onda
    Bus 008 Device 002: ID 1f3a:efe8 Onda (unverified) V972 tablet in flashing mode
    Bus 003 Device 007: ID 1f3a:efe8 Onda (unverified) V972 tablet in flashing mode
    
    root@armbian:/usr/local/src/sunxi-tools# ./fel -d3:7 ver
    AWUSBFEX soc=00001651(A20) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
    root@armbian:/usr/local/src/sunxi-tools# ./fel -d8:2 ver
    AWUSBFEX soc=00001651(A20) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000

n1tehawk, 2016-03-15:
> And BTW I don't get it how to use NextThing's -d/--dev switch correctly

Thomas, check the subsequent commits in their master branch, e.g. https://github.com/NextThingCo/sunxi-tools/commit/47bafaf91e659a12443a29c74e3138c488767d84 might be relevant. I remember looking at their stuff in December and found it a little icky, probably needs to be straightened out a bit...

n1tehawk, 2016-03-15:
Looking at their command line parsing, they require you to separate `busnum:devnum` from the `--dev` switch! They're evaluating `argv[2]`, so your `-d3:7` or `-d8:2` would never translate into something meaningful. It has to be `-d 3:7`, `-d 8:2`, ...

ThomasKaiser, 2016-03-15:
> It has to be -d 3:7, -d 8:2

Thx, that did the trick! Great, then I can proceed further...

    root@armbian:/usr/local/src/sunxi-tools# ./fel -d 3:8 ver
    AWUSBFEX soc=00001651(A20) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
    root@armbian:/usr/local/src/sunxi-tools# ./fel -d 3:7 ver
    ERROR: No USB FEL device at 0x3:0x7

n1tehawk, 2016-03-17:
@ThomasKaiser The (v5) patch linked above may be considered "final" regarding the implementation details. It would be great if you could give it some testing in an actual multi-device setup.

Regards, NiteHawk

ThomasKaiser, 2016-03-17:
@n1tehawk That looks good to me:

    root@armbian:/usr/local/src/sunxi-tools# lsusb | grep Onda
    Bus 008 Device 003: ID 1f3a:efe8 Onda (unverified) V972 tablet in flashing mode
    Bus 007 Device 002: ID 1f3a:efe8 Onda (unverified) V972 tablet in flashing mode
    Bus 003 Device 013: ID 1f3a:efe8 Onda (unverified) V972 tablet in flashing mode
    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel --dev 7:2 ver
    AWUSBFEX soc=00001651(A20) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel --dev 3:13 ver
    AWUSBFEX soc=00001680(H3) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel --dev 8:3 ver
    AWUSBFEX soc=00001673(A83T) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000

Thx! Now looking forward testing 'sunxi-fel sid' ;)

n1tehawk, 2016-03-19:
> Thx! Now looking forward testing 'sunxi-fel sid' ;)

Give https://gist.github.com/n1tehawk/71430ceab389f3167a6f a try... :wink:

@ssvb Do you think that looks sane? I can submit it as a patch to the ML, but it obviously requires your `aw_fel_readl()` to be present in the first place.

Regards, NiteHawk

n1tehawk, 2016-03-20:
@ThomasKaiser Since you seem to own that hardware, can you confirm that reading the SID (key) from address *0x01C14200* works for A83T and H3? (see https://groups.google.com/d/msg/linux-sunxi/KM--yn2B6g0/VJUmyL0xAwAJ)

Regards, NiteHawk

ThomasKaiser, 2016-03-20:
I tested 2 x H3 and 1 x A83T also (added the SID stuff there too, your patch didn't worked and I ended up with [this](http://sprunge.us/eKFM))

    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel ver sid
    AWUSBFEX soc=00001680(H3) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
    02004620 3434050c 503f0ace 44000000
    
    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel ver sid
    AWUSBFEX soc=00001680(H3) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
    02004620 6435410c 50270d0e 2c000000
    
    root@armbian:/usr/local/src/sunxi-tools# ./sunxi-fel ver sid
    AWUSBFEX soc=00001673(A83T) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
    32c00401 4020461c 00f08100 a5075604

n1tehawk, 2016-03-20:
Hmmm... Looks like at least you got some values out of it there, though they don't seem to follow the pattern we know from the A10/A13/A20 (where the first value would also contain the SoC ID).

We probably need independent confirmation by more A83T/H3 users...

Thanks for testing!

ThomasKaiser, 2016-03-20:
The first value is always 02004620 on any of the 3 H3 devices lying here around.

n1tehawk, 2016-03-20:
Yes, that might simply be a change from Allwinner over the previous e-fuses content we know from the older SoCs.

The simple fact that you get a reply for the `readl` at that address at least proves there is some (hardware) register accessible there. (Otherwise - e.g. in case of an access violation - the scratch buffer code would likely crash the FEL.)

My minimum goal would be to make sure that activating this address for A83T and H3 doesn't cause havoc when issuing a "sunxi-fel sid". So far, that seems okay.

Regards, NiteHawk

n1tehawk, 2016-03-20:
Support for the new `--dev` option has now been commited to the master branch.

n1tehawk, 2016-04-04:
I have created two more issues to follow up on this one, and am therefore closing it now.

closed 2016-04-04 by n1tehawk