linux-sunxi/sunxi-tools #39: Support a "--sid" option for device selection

n1tehawk, 2016-04-04:
We have incorporated some [support for multiple USB devices](https://github.com/linux-sunxi/sunxi-tools/commit/472ac4759df557c00248e557beb869f4fe7d75f7) connected to the same host in our `sunxi-fel` utility, allowing to select a specific device via USB bus and device number (`--dev` option).

As proposed in #37, an even more user-friendly way might be to support selection based on some unique device identifier; for example the SID. This could be done by adding a corresponding `--sid <pattern>` option. The details of how to match _pattern_ haven't been elaborated yet. (substring, regex, ...?) 

Right now the `fel.c` code is not prepared well for this enhancement. Historically, we've always created / selected a libusb handle **once** in a preparatory step, and then kept using it for an entire `sunxi-fel` "session". This needs to be rethought if we want to query several devices "in turn" (involving individual FEL communication to retrieve the ID each time), until we find the one that the user actually requested.

See also: #38

n1tehawk, 2016-04-04:
Note: Implementing ``sunxi-fel sid` should be a higher priority than this. With working SID output (e.g. usable for `grep`), some scripting might serve as a workaround to select the desired device.

This would print the SIDs of all connected FEL devices:
``` shell
#!/bin/sh
for dev in `lsusb | awk '/1f3a:efe8/ { printf "%s:%.3s ", $2, $4 }'`; do
        echo USB device $dev
        ./sunxi-fel --dev=$dev sid
done
```

ssvb, 2016-05-19:
BTW, the existing ``--dev`` option does not seem to be very useful to me because the usb id keeps changing after re-plugging the device. So it would be very useful to have the ``--sid`` option implemented. The wrapper script works but is a little bit awkward to use, so it's better to have everything nicely supported by the sunxi-fel tool itself.

n1tehawk, 2016-12-01:
Implemented in 448fa5f74c0db5c359b72ef4770a8bb84309c947.

closed 2016-12-01 by n1tehawk