linux-sunxi/sunxi-tools #48: Remove the workaround code from fel-sdboot.c (and possibly uart0-helloworld)

n1tehawk, 2016-05-14:
The root cause of the problems that surfaced with these ARM binaries is that the U-Boot `mksunxiboot` utility writes a minimal header that is **too small** to avoid corruption of actual code by the BROM bootloader. See https://patchwork.ozlabs.org/patch/622173.

Once this is fixed (in upstream U-Boot), we should probably revert the _NOP_ padding used in `fel-sdboot.c` to circumvent this. The `uart0-helloworld-sdboot.sunxi` under development currently also contains a workaround that won't be needed any more afterwards.

ssvb, 2016-05-19:
Yes, the workaround needs to be at least changed to branch over the corrupted location. The description at https://github.com/linux-sunxi/sunxi-tools/blob/ce9cf33606492076b81e1157ba9fc54b56379335/fel-sdboot.c#L41-L50 also needs to be updated. It did accurately state the facts, but now we have more information.

As for removing the workaround completely, some people may still try to recompile these binaries and use a buggy ``mksunxiboot`` tool while doing this. It might be safer to keep the workarounds at least for a year or so, maybe even longer.

n1tehawk, 2016-05-27:
Okay, i'll tag this as "_wontfix_" for the time being - so the issue can be left open until we decide that a sane `mksunxiboot` should have made it everywhere, and the workaround gets finally removed.

n1tehawk, 2016-07-25:
Note: U-Boot release _v2016.07_ contains the [related commit](http://git.denx.de/?p=u-boot.git;a=commit;h=b19236fd1c1ef289bab9e243ee5b50d658fcac3f) that fixes SPL header size, and thus `mksunxiboot` operation. Images created with this (or a later) version account for the field written by the BROM, i.e. they should no longer suffer from code corruption.

closed 2026-09-08 by paulkocialkowski