linux-sunxi/sunxi-tools #58: Please do a current release

StefanBruens, 2016-09-19:
Since the last release, a lot of changes have accumulated. Please evaluate if the current state has release quality and tag it.

Tnx.

far5893, 2016-10-21:
Quote, and add a --version flag to all commands

n1tehawk, 2016-10-22:
The current plan is to first integrate support for environment "overrides" (#60) and some smaller fixes. With those merged, we should be in a good state to tag a new release. I've also added a proposal to support version information (#62) for the various sunxi-tools utilities - comments and suggestions welcome.

Regards, NiteHawk

n1tehawk, 2016-10-26:
[sunxi-tools v1.4](https://github.com/linux-sunxi/sunxi-tools/releases/tag/v1.4) released

Note: From now on, programs should display version information in their usage help. For those binaries that do not output their usage by default (e.g. because they would process _stdin_), you may pass a `-?` option to get help, and thus version information.

Regards, NiteHawk

closed 2016-10-26 by n1tehawk