linux-sunxi/sunxi-tools #64: A33(A23) seem non supported in uart0 hello world  ?

far5893, 2016-10-24:
Hi,
I'm trying to boot A33 tablet with fel or sdcard ,
i have breakout sd adapter , can i use uart0 hello world mixing uart0 and SD?

And with fel is possible to load hello-world image to sram and execute(whiteout SD)?

n1tehawk, 2016-10-24:
Hello far5893,

Currently there is no support for A33/A23, as we're lacking the necessary hardware to test it. However, the code itself is pretty much self-explanatory, and simple enough to adjust for other SoCs. You'd mostly have to figure out the correct GPIO pin mapping (from the datasheet) and extend `gpio_init()` accordingly.

The [SD breakout](https://linux-sunxi.org/MicroSD_Breakout) might require some extra care in getting the "pinmux" (GPIO multiplexing functions) setup right. See the A13 case for an example of this.

Using an SD breakout adapter complicates things in two ways:

1. You can't use a regular (µ)SD card at the same time, as they both occupy the same slot - i.e. are mutually exclusive. That sort of rules out using _uart0-helloworld-sdboot_ via SD card, as you wouldn't be able to get at the UART lines.
2. When using the adapter, signal lines from the SoC need to get multiplexed (= reassigned) from their original "SD card" function (mmc0) to serial input/output (uart0). So even if you managed to *physically* connect both a memory card and the breakout, you could't use both functions at the same time *logically*.

That said: Yes, it's possible to upload and execute the _uart0-helloworld-sdboot.sunxi_ via USB in FEL mode, eliminating the need for / use of a memory card.

Regards, NiteHawk

Fusseldieb, 2023-07-19:
> That said: Yes, it's possible to upload and execute the uart0-helloworld-sdboot.sunxi via USB in FEL mode, eliminating the need for / use of a memory card.

How would that work? Pardon my lack of understanding.

ssvb, 2023-07-19:
Just run `sunxi-fel spl uart0-helloworld-sdboot.sunxi` on your computer (after switching the device in FEL mode and connecting a USB cable). There are wiki pages, which explain the process in more details.

closed 2026-09-08 by paulkocialkowski

miky2k, 2026-09-08:
Only 10 years to complete, nice!! I forgot what i'm doing at the time!!!