linux-sunxi/sunxi-tools #69: sunxi-tools 1.4 doesn't install sunxi-nand-image-builder

daym, 2016-10-26:
sunxi-tools 1.4 doesn't install sunxi-nand-image-builder

n1tehawk, 2016-10-26:
When the `sunxi-nand-image-builder` utility was initially submitted/discussed on the linux-sunxi mailing list, it was in fact me who suggested that it would go into *MISC_TOOLS* instead of *TOOLS* - due to the rather specific nature of this tool, and the fact that it might not be relevant / of interest to a broader audience. That also means it doesn't get built by default and/or included in `make install`.

You seem to be suggesting that this is not the case - as a consequence it should probably be moved from *MISC_TOOLS* to *TOOLS*.

@bbrezillon, @ssvb What do you think?

Regards, NiteHawk

daym, 2016-10-27:
I'm just asking. Guix usually has the policy of including all the bells and whistles so in cases like this we ask to make sure that the authors intended to leave it off. 

If you did, okay. I'm not sure why you would write a program nobody could use since it doesn't get installed - but maybe it's a debug tool.

But I checked the source code and it doesn't seem to need target /dev/mem or anything. It looks like any other image builder. So not sure why it's not just a normal tool. For example FPGA toolchains also have lots of weird programs that do things that work on just this FPGA block - they are still installed. Maybe someone needs it.

n1tehawk, 2016-10-28:
Our assumption (or lame excuse :stuck_out_tongue:) so far was that users that would actually want / benefit from the specialized tools would be tech-savvy enough to do `git clone` and `make misc`.

But I acknowledge that we probably should cater for different needs here, and leave it up to the package maintainers what selection of binaries they actually want to include in their distribution.

How about adding `make install-misc` to handle things in a way similar to what *install-tools* and *install-target-tools* already do? If your package really wants to include everything plus the kitchen sink, it could then use `make install-all install-misc` when building...

Regards, NiteHawk

daym, 2016-10-29:
Yeah, I think that's a good idea.

n1tehawk, 2016-10-29:
Done.

The various build targets are now also documented in the README, see https://github.com/linux-sunxi/sunxi-tools#building.

Regards, NiteHawk

daym, 2016-10-29:
Works for me. Thanks!

Note that sunxi-script_extractor is still not installed. I've just added it to the Guix build recipe manually for now.

n1tehawk, 2016-10-29:
That's somewhat on purpose. `sunxi-script_extractor` is a rather sloppy tool hacked together to dump configuration data (_script.bin_) from a fixed memory location. The tool has fallen into disuse / isn't actively maintained, users were only having limited success with it anyway, and IIRC it won't work on the more recent Allwinner SoCs.

daym, 2016-10-29:
Okay, thanks!

n1tehawk, 2016-10-29:
If you want it included nevertheless, you could now override _TARGET_TOOLS_ the way you initially wanted. :wink:

`make target-tools TARGET_TOOLS=sunxi-meminfo sunxi-pio sunxi-script_extractor`

ssvb, 2016-10-29:
My understanding is that the sunxi-nand-image-builder is currently a provisional work-in-progress code dump. It is not expected to be useful for the end users yet. At least not until the NAND support is in a better shape in the mainline Linux and U-Boot. And it may take months (or years?) before this happens. The https://linux-sunxi.org/Mainline_NAND_Howto wiki page provides some additional details (note that there is even no reference to the sunxi-nand-image-builder tool there).

bbrezillon, 2016-10-29:
@ssvb: then you did not understand what I tried to explain when I submitted this tool. sunxi-nand-image-builder will always be useful and is not meant to be replaced by something else. I know you'd like a more user-friendly solution, but there's nothing like that right now, and, AFAIK, nobody is working on such a solution.
Also note that this tool is completely independent of the u-boot and linux implementation, it's just there to generate raw NAND images, which is required if you want to create an SPL image bootable from NAND.

Regarding u-boot and linux support for the sunxi NAND controller, it's already there and functional, the only thing we miss is reliable support for MLC NANDs, which is not Allwinner specific, but is impacting all boards using an MLC NAND.

Yes, I know, the wiki page is not up-to-date, and it's partly because I'm lazy. But the other good reason is that I don't want people to start using the NAND on boards which embed an MLC chip (which, AFAIK, is the case of all Allwinner based devices embedding a NAND, except the CHIP Pro).

So please, stop considering sunxi-nand-image-builder as my personal development/debugging tool. It's been used by other people which were trying to generate a SPL image bootable from NAND. And even if you don't like the current solution, please accept that there's currently no other alternative.

bbrezillon, 2016-10-29:
@n1tehawk: I'm perfectly fine with the install-misc solution you proposed. Thanks.

closed 2016-11-13 by n1tehawk

ssvb, 2017-08-15:
@bbrezillon: I don't care about what you say. What really matters is what you do. The "experimental work-in progress code" was (and still is) a rather accurate description. Hey, I did not label it as "unusable piece of crap" or something like this, so don't feel offended. Yes, you are lazy and I'm neither your boss nor paying you for doing this work. Still dumping an unfinished piece of code on us was not very nice either, because it becomes a maintenance burden for the others. And it also causes confusion for the end users and distro maintainers (see the question of the original reporter).

I questioned the usefulness of having sunxi-nand-image-builder in sunxi-tools during the review of your patch because it was never a part of a documented and usable solution. So nobody could actually test it (at least not without spending an unreasonable amount of time on figuring out undocumented bits and pieces). And if you did not want people to start using NAND, then you did not have to submit your  program to sunxi-tools in the first place.

Nowadays sunxi-nand-image-builder is maintained as a part of U-Boot, you can find it at http://git.denx.de/?p=u-boot.git;a=blob;f=tools/sunxi-spl-image-builder.c
It is also now documented at http://git.denx.de/?p=u-boot.git;a=blob;f=board/sunxi/README.nand (far from perfect, but better than nothing). So I guess, now it's best to remove sunxi-nand-image-builder from sunxi-tools and let U-Boot people take care of it.

bbrezillon, 2017-08-15:
@ssvb: do whatever you want, I don't care anymore. And I certainly won't spend time having this discussion with you.

You say I'm lazy, maybe you're right. But did you ever ask yourself if you're doing good at driving the sunxi community? I think you're not. Actually, I think you're just a poison to the sunxi community. You're here saying what people should do or work on and just reject anything that's not exactly like you want.

What's for sure is that I'll never contribute to the sunxi-tools project (or any project you maintain) again.

ssvb, 2017-08-15:
@bbrezillon I merely assessed the outcome of our previous discussion since we got a new development. And it turned out that I was right all along. There was no need for having sunxi-nand-image-builder in sunxi-tools. And it should not have been pushed to the repository in the first place since it did not pass the basic acceptance criteria (having documentation and usage instructions is one of them). We did bend the rules a little bit with a hope that you would keep working on improving NAND support, but nothing good came out of it.

Has anyone told you that you are very rude? You will not make many friends with such an attitude.

BTW, I'm not maintaining sunxi-tools, but @n1tehawk does (or at least did a while ago). And who told you that I'm driving the sunxi community or even trying to drive it?