linux-sunxi/sunxi-tools #74: 1.4.1 fails building fel with gcc 6.2.1 on Fedora

nullr0ute, 2016-10-29:
Lots of undefined errors

<pre>
+ make -j4 'CFLAGS=-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic -Iinclude -D_POSIX_C_SOURCE=200112L -std=c99'
./autoversion.sh: line 9: git: command not found
Unable to determine current version (using "v1.4.1" as fallback)
cc -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic -Iinclude -D_POSIX_C_SOURCE=200112L -std=c99  -o sunxi-fexc fexc.c script.c script_uboot.c script_bin.c script_fex.c 
cc -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic -Iinclude -D_POSIX_C_SOURCE=200112L -std=c99  -o sunxi-bootinfo bootinfo.c 
cc -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic -Iinclude -D_POSIX_C_SOURCE=200112L -std=c99 `pkg-config --cflags libusb-1.0`  -o sunxi-fel fel.c progress.c  `pkg-config --libs libusb-1.0`
cc -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic -Iinclude -D_POSIX_C_SOURCE=200112L -std=c99 -c -o nand-part-main.o nand-part-main.c
fel.c: In function 'get_image_type':
fel.c:164:6: warning: implicit declaration of function 'be32toh' [-Wimplicit-function-declaration]
  if (be32toh(buf32[0]) != IH_MAGIC) /* signature mismatch */
      ^~~~~~~
fel.c: In function 'aw_send_usb_request':
fel.c:178:14: warning: implicit declaration of function 'htole16' [-Wimplicit-function-declaration]
   .request = htole16(type),
              ^~~~~~~
fel.c:179:13: warning: implicit declaration of function 'htole32' [-Wimplicit-function-declaration]
   .length = htole32(length),
             ^~~~~~~
fel.c: In function 'aw_fel_get_version':
fel.c:242:17: warning: implicit declaration of function 'le32toh' [-Wimplicit-function-declaration]
  buf->soc_id = (le32toh(buf->soc_id) >> 8) & 0xFFFF;
                 ^~~~~~~
fel.c:245:20: warning: implicit declaration of function 'le16toh' [-Wimplicit-function-declaration]
  buf->scratchpad = le16toh(buf->scratchpad);
                    ^~~~~~~
fel.c: In function 'aw_fel_write_and_execute_spl':
fel.c:1217:2: warning: implicit declaration of function 'usleep' [-Wimplicit-function-declaration]
  usleep(250000);
  ^~~~~~
cc -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic -Iinclude -D_POSIX_C_SOURCE=200112L -std=c99 -c -o nand-part-a10.o nand-part.c -D A10
nand-part.c: In function 'writembrs':
nand-part.c:240:2: warning: ignoring return value of 'read', declared with attribute warn_unused_result [-Wunused-result]
  read(0, &yn, 1);
  ^~~~~~~~~~~~~~~
nand-part.c:251:3: warning: ignoring return value of 'write', declared with attribute warn_unused_result [-Wunused-result]
   write(fd,mbr,MBR_SIZE);
   ^~~~~~~~~~~~~~~~~~~~~~
cc -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic -Iinclude -D_POSIX_C_SOURCE=200112L -std=c99 -c -o nand-part-a20.o nand-part.c -D A20
nand-part.c: In function 'writembrs':
nand-part.c:240:2: warning: ignoring return value of 'read', declared with attribute warn_unused_result [-Wunused-result]
  read(0, &yn, 1);
  ^~~~~~~~~~~~~~~
nand-part.c:251:3: warning: ignoring return value of 'write', declared with attribute warn_unused_result [-Wunused-result]
   write(fd,mbr,MBR_SIZE);
   ^~~~~~~~~~~~~~~~~~~~~~
cc  -o sunxi-nand-part nand-part-main.o nand-part-a10.o nand-part-a20.o 
ln -nsf sunxi-fexc bin2fex
ln -nsf sunxi-fexc fex2bin
/tmp/ccgEW415.o: In function `get_image_type':
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:164: undefined reference to `be32toh'
/tmp/ccgEW415.o: In function `aw_send_usb_request':
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:179: undefined reference to `htole32'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:180: undefined reference to `htole32'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:178: undefined reference to `htole16'
/tmp/ccgEW415.o: In function `aw_send_fel_request':
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:223: undefined reference to `htole32'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:224: undefined reference to `htole32'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:225: undefined reference to `htole32'
/tmp/ccgEW415.o: In function `aw_fel_get_version':
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:242: undefined reference to `le32toh'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:243: undefined reference to `le32toh'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:244: undefined reference to `le32toh'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:245: undefined reference to `le16toh'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:246: undefined reference to `le32toh'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:247: undefined reference to `le32toh'
/tmp/ccgEW415.o: In function `aw_fel_write_uboot_image':
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1267: undefined reference to `be32toh'
[... 38 lines cut ...]
/tmp/ccgEW415.o:/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:948: more undefined references to `htole32' follow
/tmp/ccgEW415.o: In function `aw_backup_and_disable_mmu':
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:996: undefined reference to `le32toh'
/tmp/ccgEW415.o: In function `aw_restore_and_enable_mmu':
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1027: undefined reference to `htole32'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1028: undefined reference to `htole32'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1029: undefined reference to `htole32'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1030: undefined reference to `htole32'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1031: undefined reference to `htole32'
/tmp/ccgEW415.o:/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1032: more undefined references to `htole32' follow
/tmp/ccgEW415.o: In function `aw_fel_write_and_execute_spl':
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1102: undefined reference to `le32toh'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1103: undefined reference to `le32toh'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1112: undefined reference to `le32toh'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1207: undefined reference to `htole32'
/tmp/ccgEW415.o: In function `pass_fel_information':
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1367: undefined reference to `htole32'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1368: undefined reference to `htole32'
/tmp/ccgEW415.o: In function `aw_rmr_request':
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1436: undefined reference to `htole32'
/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1437: undefined reference to `htole32'
/tmp/ccgEW415.o:/builddir/build/BUILD/sunxi-tools-1.4.1/fel.c:1438: more undefined references to `htole32' follow
collect2: error: ld returned 1 exit status
make: *** [Makefile:119: sunxi-fel] Error 1
</pre>

n1tehawk, 2016-10-29:
You're overriding `CFLAGS`, killing the `DEFINES` that would normally get passed from our _Makefile_, e.g. https://github.com/linux-sunxi/sunxi-tools/blob/v1.4.1/Makefile#L25-L27. The absence of `-D_BSD_SOURCE` then cripples _portable_endian.h_.

If you wish to pass **additional** CFLAGS, you should preferable use `EXTRA_CFLAGS` instead.

Regards, NiteHawk

ssvb, 2016-10-29:
I would say that we should use CFLAGS for passing additional options (instead of inventing EXTRA_CFLAGS), because it's a common practice used by all build systems. Or at least by autotools and CMake.

n1tehawk, 2016-10-29:
How do we ensure we get our required flags in? Add whatever we need on top of existing CFLAGS - e.g. `CFLAGS += -D_BSD_SOURCE` -, or pass them independently in the Makefile rules?

ssvb, 2016-10-29:
Maybe introduce a new DEFAULT_CFLAGS variable in the makefile? Then append the user supplied CFLAGS variable to it. The user's CFLAGS should have the highest priority and the power to override the default options. Again, this is the way how the other build systems work.

n1tehawk, 2016-10-29:
Yes, that seems reasonable.

ssvb, 2016-10-29:
Some information about CFLAGS: https://www.gnu.org/software/automake/manual/html_node/User-Variables.html

closed 2016-11-13 by n1tehawk