linux-sunxi/sunxi-tools #76: sunxi-meminfo can not be built on aarch64

StefanBruens, 2016-10-29:
meminfo.c tries to include <sys/io.h>, which is not available on aarch64, see
https://bugzilla.redhat.com/show_bug.cgi?id=1116162

As sunxi-meminfo and sunxi-pio are both part of target-tools, it is difficult to build/install only sunxi-pio

n1tehawk, 2016-10-30:
Hi Stefan!

A possible workaround could be to narrow down on _target-tools_ by overriding the selection of binaries:
`make install-target-tools TARGET_TOOLS=sunxi-pio`
Admittedly this is somewhat awkward.

The AArch64 problem seems a bit tricky to tackle with the current 'static' Makefile approach. My impression is that it would require some preprocessing with regard to the cross-toolchain, i.e. either test if `<sys/io.h>` is available, or whether the target architecture is affected (e.g. by parsing the output of `$(CROSS_CC) -v`?). We're probably heading for _autotools_ territory here - or do you know of a feasible alternative approach?

Regards, NiteHawk

n1tehawk, 2016-11-04:
Is it actually sensible to try building `target-tools` using AArch64? From what I've learned so far, the 'native' compiler would be unable to generate the AArch32 code we require. You'd have to cross-compile using _arm-linux-gnueabi[hf]_ anyway...

n1tehawk, 2016-11-21:
Closed due to lack of feedback.

closed 2016-11-21 by n1tehawk