linux-sunxi/sunxi-tools #84: script-extractor on A33 fail

far5893, 2016-11-21:
Semms that address offest is wrong 
script extractor first bytes :  
```

00000000  4f 00 00 00 74 a8 00 00  01 00 00 00 02 00 00 00  |O...t...........|
00000010  70 72 6f 64 75 63 74 00  00 00 00 00 00 00 00 00  |product.........|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  02 00 00 00 1a 03 00 00  70 6c 61 74 66 6f 72 6d  |........platform|
00000040  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000050  00 00 00 00 00 00 00 00  03 00 00 00 2e 03 00 00  |................|
00000060  74 61 72 67 65 74 00 00  00 00 00 00 00 00 00 00  |target..........|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000080  05 00 00 00 4c 03 00 00  6b 65 79 5f 64 65 74 65  |....L...key_dete|
00000090  63 74 5f 65 6e 00 00 00  00 00 00 00 00 00 00 00  |ct_en...........|
000000a0  00 00 00 00 00 00 00 00  01 00 00 00 7e 03 00 00  |............~...|
000000b0  70 6f 77 65 72 5f 73 70  6c 79 00 00 00 00 00 00  |power_sply......|
000000c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000d0  07 00 00 00 88 03 00 00  63 61 72 64 5f 62 6f 6f  |........card_boo|
000000e0  74 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |t...............|
000000f0  00 00 00 00 00 00 00 00  03 00 00 00 ce 03 00 00  |................|
00000100  63 61 72 64 30 5f 62 6f  6f 74 5f 70 61 72 61 00  |card0_boot_para.|
00000110  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000120  09 00 00 00 ec 03 00 00  63 61 72 64 32 5f 62 6f  |........card2_bo|
00000130  6f 74 5f 70 61 72 61 00  00 00 00 00 00 00 00 00  |ot_para.........|
00000140  00 00 00 00 00 00 00 00  0a 00 00 00 46 04 00 00  |............F...|
00000150  74 77 69 5f 70 61 72 61  00 00 00 00 00 00 00 00  |twi_para........|
00000160  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000170  03 00 00 00 aa 04 00 00  75 61 72 74 5f 70 61 72  |........uart_par|
00000180  61 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |a...............|
00000190  00 00 00 00 00 00 00 00  03 00 00 00 c8 04 00 00  |................|
000001a0  66 6f 72 63 65 5f 75 61  72 74 5f 70 61 72 61 00  |force_uart_para.|
000001b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000001c0  03 00 00 00 e6 04 00 00  6a 74 61 67 5f 70 61 72  |........jtag_par|
000001d0  61 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |a...............|
000001e0  00 00 00 00 00 00 00 00  05 00 00 00 04 05 00 00  |................|
000001f0  63 6c 6f 63 6b 00 00 00  00 00 00 00 00 00 00 00  |clock...........|
00000200  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000210  0f 00 00 00 36 05 00 00  70 6d 5f 70 61 72 61 00  |....6...pm_para.|
00000220  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000230  00 00 00 00 00 00 00 00  01 00 00 00 cc 05 00 00  |................|
00000240  64 72 61 6d 5f 70 61 72  61 00 00 00 00 00 00 00  |dram_para.......|
00000250  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000260  18 00 00 00 d6 05 00 00  70 6d 5f 70 61 72 61 00  |........pm_para.|
00000270  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000280  00 00 00 00 00 00 00 00  01 00 00 00 c6 06 00 00  |................|
00000290  77 61 6b 65 75 70 5f 73  72 63 5f 70 61 72 61 00  |wakeup_src_para.|
000002a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000002b0  07 00 00 00 d0 06 00 00  74 77 69 30 00 00 00 00  |........twi0....|

```

n1tehawk, 2016-11-21:
I've mentioned this before in https://github.com/linux-sunxi/sunxi-tools/issues/69#issuecomment-257085815.

The script extractor is an extremely (overly?) simple program that just dumps a specific region of SoC memory, namely `SCRIPT_SIZE` (0x20000) bytes from address `SCRIPT_START` (0x43000000). The address may be subject to change - and AFAICT its current (hardcoded) value was chosen from older U-Boot, so it may no longer apply in your case.

We should probably try to make this tools somewhat 'smarter' by using some heuristics, and possibly use a memory scan to detect the start of _script.bin_-compatible data in RAM.

far5893, 2016-11-21:
This are first bytes of  a working bin2fex file ,
as you can see offset is correct but header is different , is a problem of bin2fex ?

00000000  4d 00 00 00 00 00 00 00  01 00 00 00 02 00 00 00  |M...............|
00000010  70 72 6f 64 75 63 74 00  00 00 00 00 00 00 00 00  |product.........|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  02 00 00 00 06 03 00 00  70 6c 61 74 66 6f 72 6d  |........platform|
00000040  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000050  00 00 00 00 00 00 00 00  03 00 00 00 1a 03 00 00  |................|
00000060  74 61 72 67 65 74 00 00  00 00 00 00 00 00 00 00  |target..........|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000080  05 00 00 00 38 03 00 00  6b 65 79 5f 64 65 74 65  |....8...key_dete|
00000090  63 74 5f 65 6e 00 00 00  00 00 00 00 00 00 00 00  |ct_en...........|
000000a0  00 00 00 00 00 00 00 00  01 00 00 00 6a 03 00 00  |............j...|
000000b0  70 6f 77 65 72 5f 73 70  6c 79 00 00 00 00 00 00  |power_sply......|
000000c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000d0  07 00 00 00 74 03 00 00  63 61 72 64 5f 62 6f 6f  |....t...card_boo|
000000e0  74 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |t...............|
000000f0  00 00 00 00 00 00 00 00  03 00 00 00 ba 03 00 00  |................|
00000100  63 61 72 64 30 5f 62 6f  6f 74 5f 70 61 72 61 00  |card0_boot_para.|
00000110  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000120  09 00 00 00 d8 03 00 00  63 61 72 64 32 5f 62 6f  |........card2_bo|
00000130  6f 74 5f 70 61 72 61 00  00 00 00 00 00 00 00 00  |ot_para.........|
00000140  00 00 00 00 00 00 00 00  0a 00 00 00 32 04 00 00  |............2...|
00000150  74 77 69 5f 70 61 72 61  00 00 00 00 00 00 00 00  |twi_para........|
00000160  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000170  03 00 00 00 96 04 00 00  75 61 72 74 5f 70 61 72  |........uart_par|
00000180  61 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |a...............|
00000190  00 00 00 00 00 00 00 00  03 00 00 00 b4 04 00 00  |................|
000001a0  66 6f 72 63 65 5f 75 61  72 74 5f 70 61 72 61 00  |force_uart_para.|
000001b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000001c0  03 00 00 00 d2 04 00 00  6a 74 61 67 5f 70 61 72  |........jtag_par|
000001d0  61 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |a...............|
000001e0  00 00 00 00 00 00 00 00  05 00 00 00 f0 04 00 00  |................|
000001f0  63 6c 6f 63 6b 00 00 00  00 00 00 00 00 00 00 00  |clock...........|
00000200  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000210  0d 00 00 00 22 05 00 00  70 6d 5f 70 61 72 61 00  |...."...pm_para.|
00000220  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000230  00 00 00 00 00 00 00 00  01 00 00 00 a4 05 00 00  |................|
00000240  64 72 61 6d 5f 70 61 72  61 00 00 00 00 00 00 00  |dram_para.......|
00000250  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000260  18 00 00 00 ae 05 00 00  70 6d 5f 70 61 72 61 00  |........pm_para.|
00000270  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000280  00 00 00 00 00 00 00 00  01 00 00 00 9e 06 00 00  |................|
00000290  77 61 6b 65 75 70 5f 73  72 63 5f 70 61 72 61 00  |wakeup_src_para.|
000002a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000002b0  07 00 00 00 a8 06 00 00  74 77 69 30 00 00 00 00  |........twi0....|
000002c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000002d0  00 00 00 00 00 00 00 00  03 00 00 00 ee 06 00 00  |................|
000002e0  74 77 69 31 00 00 00 00  00 00 00 00 00 00 00 00  |twi1............|
000002f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000300  03 00 00 00 0c 07 00 00  74 77 69 32 00 00 00 00  |........twi2....|

n1tehawk, 2016-11-21:
>because bin2fex give a Malformed error.

Care to tell us the exact error message, and the version output of `./bin2fex -?`?

n1tehawk, 2016-11-21:
I have a suspicion that you're **not** using the latest _HEAD_ version from the repository, so you're probably lacking commit 2a4af4f5 - that would explain the error. Could you retry `bin2fex` built from a fresh clone/checkout?

far5893, 2016-11-22:
sunxi-fexc v1.4.1  works ok.
Sorry i used a month old version...you can close

closed 2016-11-22 by n1tehawk