linux-sunxi/sunxi-tools #89: Doesn't compile on NetBSD 7

zilti, 2016-12-17:
```
cc -std=c99 -Wall -Wextra -Wno-unused-result -D_POSIX_C_SOURCE=200112L -D_BSD_SOURCE -D_DEFAULT_SOURCE -Iinclude/ -I/usr/include/ -I/
usr/pkg/include/  `pkg-config --cflags libusb-1.0`  -o sunxi-fel fel.c progress.c soc_info.c fel_lib.c  `pkg-config --libs libusb-1.0
`
fel.c: In function 'get_image_type':
fel.c:70:2: warning: implicit declaration of function 'be32toh' [-Wimplicit-function-declaration]
  if (be32toh(buf32[0]) != IH_MAGIC) /* signature mismatch */
  ^
fel.c: In function 'aw_read_arm_cp_reg':
fel.c:235:3: warning: implicit declaration of function 'htole32' [-Wimplicit-function-declaration]
   htole32(opcode),     /* mrc  coproc, opc1, r0, crn, crm, opc2 */
   ^
fel.c:242:2: warning: implicit declaration of function 'le32toh' [-Wimplicit-function-declaration]
  return le32toh(val);
  ^
fel_lib.c: In function 'aw_send_usb_request':
fel_lib.c:142:3: warning: implicit declaration of function 'htole16' [-Wimplicit-function-declaration]
   .request = htole16(type),
   ^
fel_lib.c:143:3: warning: implicit declaration of function 'htole32' [-Wimplicit-function-declaration]
   .length = htole32(length),
   ^
fel_lib.c: In function 'aw_fel_get_version':
fel_lib.c:200:2: warning: implicit declaration of function 'le32toh' [-Wimplicit-function-declaration]
  buf->soc_id = (le32toh(buf->soc_id) >> 8) & 0xFFFF;
  ^
fel_lib.c:203:2: warning: implicit declaration of function 'le16toh' [-Wimplicit-function-declaration]
  buf->scratchpad = le16toh(buf->scratchpad);
  ^
/var/tmp//ccoHTHJa.o: In function `get_image_type':
fel.c:(.text+0x34): undefined reference to `be32toh'
/var/tmp//ccoHTHJa.o: In function `aw_read_arm_cp_reg':
fel.c:(.text+0x813): undefined reference to `htole32'
fel.c:(.text+0x825): undefined reference to `htole32'
fel.c:(.text+0x837): undefined reference to `htole32'
fel.c:(.text+0x899): undefined reference to `le32toh'
/var/tmp//ccoHTHJa.o: In function `aw_write_arm_cp_reg':
fel.c:(.text+0x904): undefined reference to `htole32'
fel.c:(.text+0x916): undefined reference to `htole32'
fel.c:(.text+0x928): undefined reference to `htole32'
fel.c:(.text+0x93a): undefined reference to `htole32'
fel.c:(.text+0x94c): undefined reference to `htole32'
/var/tmp//ccoHTHJa.o:fel.c:(.text+0x95e): more undefined references to `htole32' follow
/var/tmp//ccoHTHJa.o: In function `aw_get_stackinfo':
fel.c:(.text+0xc5b): undefined reference to `le32toh'
fel.c:(.text+0xc72): undefined reference to `le32toh'
/var/tmp//ccoHTHJa.o: In function `aw_backup_and_disable_mmu':
/var/tmp//ccoHTHJa.o: In function `aw_backup_and_disable_mmu':                                                               [0/1728]
fel.c:(.text+0xf45): undefined reference to `htole32'
fel.c:(.text+0xf57): undefined reference to `htole32'
fel.c:(.text+0xf69): undefined reference to `htole32'
fel.c:(.text+0xf7b): undefined reference to `htole32'
fel.c:(.text+0xf8d): undefined reference to `htole32'
fel.c:(.text+0x1148): undefined reference to `le32toh'
/var/tmp//ccoHTHJa.o: In function `aw_restore_and_enable_mmu':
fel.c:(.text+0x12b3): undefined reference to `htole32'
fel.c:(.text+0x12c5): undefined reference to `htole32'
fel.c:(.text+0x12d7): undefined reference to `htole32'
fel.c:(.text+0x12e9): undefined reference to `htole32'
fel.c:(.text+0x12fb): undefined reference to `htole32'
/var/tmp//ccoHTHJa.o:fel.c:(.text+0x130d): more undefined references to `htole32' follow
/var/tmp//ccoHTHJa.o: In function `aw_fel_write_and_execute_spl':
fel.c:(.text+0x1627): undefined reference to `le32toh'
fel.c:(.text+0x1647): undefined reference to `le32toh'
fel.c:(.text+0x16b8): undefined reference to `le32toh'
fel.c:(.text+0x1c05): undefined reference to `htole32'
/var/tmp//ccoHTHJa.o: In function `aw_fel_write_uboot_image':
fel.c:(.text+0x1e2f): undefined reference to `be32toh'
fel.c:(.text+0x1e48): undefined reference to `be32toh'
/var/tmp//ccoHTHJa.o: In function `pass_fel_information':
fel.c:(.text+0x20ba): undefined reference to `htole32'
fel.c:(.text+0x20cc): undefined reference to `htole32'
/var/tmp//ccoHTHJa.o: In function `aw_rmr_request':
fel.c:(.text+0x2158): undefined reference to `htole32'
fel.c:(.text+0x216a): undefined reference to `htole32'
fel.c:(.text+0x217c): undefined reference to `htole32'
/var/tmp//ccoHTHJa.o:fel.c:(.text+0x218e): more undefined references to `htole32' follow
/var/tmp//ccFieRE7.o: In function `aw_send_usb_request':
fel_lib.c:(.text+0x20a): undefined reference to `htole16'
/var/tmp//ccFieRE7.o: In function `aw_send_fel_request':
fel_lib.c:(.text+0x3a2): undefined reference to `htole32'
fel_lib.c:(.text+0x3b4): undefined reference to `htole32'
fel_lib.c:(.text+0x3c6): undefined reference to `htole32'
/var/tmp//ccFieRE7.o: In function `aw_fel_get_version':
fel_lib.c:(.text+0x46d): undefined reference to `le32toh'
fel_lib.c:(.text+0x48d): undefined reference to `le32toh'
fel_lib.c:(.text+0x4ad): undefined reference to `le32toh'
fel_lib.c:(.text+0x4ca): undefined reference to `le16toh'
fel_lib.c:(.text+0x4e6): undefined reference to `le32toh'
fel_lib.c:(.text+0x502): undefined reference to `le32toh'
/var/tmp//ccFieRE7.o: In function `aw_fel_readl_n':
fel_lib.c:(.text+0x6db): undefined reference to `htole32'
fel_lib.c:(.text+0x6ed): undefined reference to `htole32'
fel_lib.c:(.text+0x6ff): undefined reference to `htole32'
fel_lib.c:(.text+0x711): undefined reference to `htole32'
fel_lib.c:(.text+0x723): undefined reference to `htole32'
/var/tmp//ccFieRE7.o:fel_lib.c:(.text+0x735): more undefined references to `htole32' follow
/var/tmp//ccFieRE7.o: In function `aw_fel_readl_n':
fel_lib.c:(.text+0x8b2): undefined reference to `le32toh'
/var/tmp//ccFieRE7.o: In function `aw_fel_writel_n':
fel_lib.c:(.text+0x9e5): undefined reference to `htole32'
fel_lib.c:(.text+0x9fa): undefined reference to `htole32'
fel_lib.c:(.text+0xa0f): undefined reference to `htole32'
fel_lib.c:(.text+0xa24): undefined reference to `htole32'
fel_lib.c:(.text+0xa39): undefined reference to `htole32'
/var/tmp//ccFieRE7.o:fel_lib.c:(.text+0xa4e): more undefined references to `htole32' follow
Makefile:129: recipe for target 'sunxi-fel' failed
gmake: *** [sunxi-fel] Error 1
```

n1tehawk, 2016-12-18:
Hi @zilti!

I think that NetBSD compilation requires an explicit `_NETBSD_SOURCE` definition for the feature test macros in _sys/endian.h_, which in turn get used by _portable_endian.h_ (see commit 805d2a8). Our Makefile currently isn't smart enough to detect NetBSD on its own, and we probably need to improve on that...

Could you check whether `make OS=NetBSD` would allow you to progress further with the build operation?

Regards, NiteHawk

n1tehawk, 2016-12-28:
Commit 1e219c0 is an attempt to improve on this. Apart from that, closed due to lack of feedback,

closed 2016-12-28 by n1tehawk

zilti, 2016-12-31:
Just wanted to add: it's in the pkgsrc repositories as a working version, I guess they modified the Makefile for that.

n1tehawk, 2017-01-01:
Thanks, that is a useful hint! It looks like _pkgsrc_ applies a series of patches (http://ftp.netbsd.org/pub/pkgsrc/current/pkgsrc/sysutils/sunxi-tools/patches/), among them a definition of `_XOPEN_SOURCE`.