linux-sunxi/sunxi-tools #97: 900kB/s FEL transfer speed on A13

kotc, 2017-02-13:
when fiddling with nolimbook i've got 570kB/s while using win7_64 laptop (using binaries provided) and 944kB/s while using linux on opipc (armbian, kernel 3.4.112+ (from armbian but my own config))

n1tehawk, 2017-02-13:
I need to verify this with my A20. I suspect the Windows `libusb` layer might introduce additional overhead, which could have a negative impact on performance. We probably can't expect the results to be similar between the two very different OS.

ssvb, 2017-02-13:
It's not just Windows. I never got transfer speeds higher than 600 kB/s on any A13 device and I'm only using Linux on my desktop computer.

What @kotc has demonstrated is that the remaining FEL transfer speed bottleneck likely depends on something in the USB stack in the host computer rather than on the device itself (the device specific bottlenecks such as the [AHB and CPU clock speeds](http://lists.denx.de/pipermail/u-boot/2015-November/234995.html), [write-combine mapping for the DRAM](https://github.com/linux-sunxi/sunxi-tools/commit/e4b3da2b17ee9d7c5cab9b80e708b3a309fc4c96) and other things have been already resolved).

wens, 2020-10-30:
No activity. Closing it.

closed 2020-10-30 by wens