linux-sunxi/sunxi-tools #100: FEL mode hangs on writing on BananaPi M2 Ultra

daym, 2017-03-19:
```
$ sudo sunxi-fel version
AWUSBFEX soc=00001701(R40) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
$ sudo sunxi-fel -p spl sunxi 
usb_bulk_send() ERROR -7: Operation timed out
$ sudo sunxi-fel version
usb_bulk_send() ERROR -7: Operation timed out
```

I then disconnect and reconnect the USB cable.

```
$ sudo sunxi-fel version
AWUSBFEX soc=00001701(R40) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 
$ sudo sunxi-fel dump 0 1000000 > Q
$ ls -l Q
-rw-r--r-- 1 dannym users 1000000 Mar 19 17:13 Q
$ sudo sunxi-fel fill 0x47000000 1000 0x42
usb_bulk_send() ERROR -7: Operation timed out
$ sudo sunxi-fel clear 0x47000000 1000
usb_bulk_send() ERROR -7: Operation timed out
```

Do I have to press some button on the device? Because I didn't.

ssvb, 2017-03-19:
What kind of 'sunxi' file are you using? Is it a U-Boot bootloader? You can also add '-v' option to the command line for more verbose reporting.

As for the "sunxi-fel fill 0x47000000 1000 0x42" command, it is expected to fail because the DRAM is not initialized yet at that point. We had an idea to add a sanity check and print a comprehensive error message in this case, but nobody has implemented it yet.

daym, 2017-03-19:
It's the U-Boot bootloader that I got from the SD card with skip=8 ...

ssvb, 2017-03-19:
> It's the U-Boot bootloader that I got from the SD card with skip=8 bs=1024; the first few bytes of it contain "eGON.BT0" without quotes.

Do you have a log from running "sunxi-fel -v spl sunxi"?

> Why does dumping work then? Is that all SRAM?

You can check the memory map in the R40 user manual. SRAM should work in sunxi-fel, but DRAM will not work unless it is initialized.

> Could you give me some commands to test it with?

As for some basic testing, you can try to run "sunxi-fel -v spl uart0-helloworld-sdboot.sunxi" first and check if it can print something on the UART serial console.

daym, 2017-03-19:
>Do you have a log from running "sunxi-fel -v spl sunxi"?

```
Stack pointers: sp_irq=0x00002000, sp=0x00005E08
MMU is not enabled by BROM
usb_bulk_send() ERROR -7: Operation timed out
```

ssvb, 2017-03-19:
Are you by chance running it in a virtual machine? There is #98 about this.

daym, 2017-03-19:
No, it's native on an X200.

ssvb, 2017-03-19:
You can try to debug what's wrong with uart0-helloworld-sdboot.sunxi first. Some time ago @wens added R40 support to sunxi-tools via #86.

Are you sure that your UART serial console works properly? Does it print anything when you are booting the stock firmware from the SD card?

daym, 2017-03-19:
I tried using sunxi-tools master's sunxi-fel from git to do the spl helloworld - it works!

The U-Boot still doesn't, though. That might be because it's not mainline uboot (I think) but the version that sinovoip had.

>Are you sure that your UART serial console works properly? 

Yes. I also made sure not to close the client application when reconnecting without sd so that it indeed continues working.

>Does it print anything when you are booting the stock firmware from the SD card?

Yes.

ssvb, 2017-03-19:
> I tried using sunxi-tools master's sunxi-fel from git to do the spl helloworld - it works!

OK, it's good to know that at least uart0-helloworld-sdboot.sunxi

> The U-Boot still doesn't, though. That might be because it's not mainline uboot (I think) but the version that sinovoip had.

Yes, the non-mainline U-Boot does not support FEL boot. Still it should be started correctly, print something to the serial console and attempt to continue booting from the SD card rather than returning control back to sunxi-fel.

daym, 2017-03-19:
Trying Chen-Yu's U-Boot (r40 branch):

```
U-Boot SPL 2017.03-rc3-g440d8ae (Mar 19 2017 - 18:39:50)
DRAM: 2048 MiB
Trying to boot from FEL
```

ssvb, 2017-03-19:
The "Trying to boot from FEL" message means that the control is supposed to be transferred back to the FEL handler in the BROM. And sunxi-fel should be able to execute commands to read/write DRAM and other things.

You can run the "sunxi-fel uboot" command instead of "sunxi-fel spl".

daym, 2017-03-19:
>And sunxi-fel should be able to execute commands to read/write DRAM and other things.

Yes, that indeed works. Also DRAM.

>You can run the "sunxi-fel uboot" command instead of "sunxi-fel spl".

Yep, works, too.

ssvb, 2017-03-19:
Excellent! I guess, everything is resolved then.

> It might make sense to release a version with these fixes...

Which fixes do you mean? The last released version of sunxi-tools just does not support Allwinner R40 yet. Though I wonder why did it identify the SoC type correctly in your original report?

daym, 2017-03-19:
Yeah, sorry for the noise.

>Which fixes do you mean?

R40 fel not timing out.

>The last released version of sunxi-tools just does not support Allwinner R40 yet. 

Oh sorry. 

>Though I wonder why did it identify the SoC type correctly in your original report?

I don't know. I checked now which it is, it's the github tarball from 1.4.2 *shrugs*...

ssvb, 2017-03-19:
I have just checked the v1.4.2 tag and it actually seems to have some support for R40. Could you please investigate what's wrong with it? Maybe by doing a quick bisecting. I don't have any R40 hardware myself. Thanks!

daym, 2017-03-19:
Sure. I've checked a diff between v1.4.2 and master and it seems usb handling in fel got refactored entirely and is now in fel_lib.c .

daym, 2017-03-19:
If I try to execute the `bin/uart0-helloworld-sdboot.sunxi` from v1.4.2 it apparently never works (also when I try to execute the old `bin/uart0-helloworld-sdboot.sunxi` from v1.4.2 using master fel).

ssvb, 2017-03-20:
> So I'd say that v1.4.2's bin/uart0-helloworld-sdboot.sunxi binary is somehow broken.

I think that's because the uart0-helloworld-sdboot.sunxi binary was only refreshed in 382bb4c2ebae30de8abd9f98ddaeb942bc794abf and the v1.4.2 has an old version of it (while it still can be compiled from sources).

> The sinovoip uboot never works via fel - no matter which fel I use.

I have no idea why it does not even start booting. In theory it should, and older boot0 releases for older SoCs worked. But if the mainline U-Boot with works (with the patches from @wens), then we probably don't need to worry.

Thanks for providing all this information!

ssvb, 2017-03-20:
> Would it be possible to make sunxi-fel distinguish between "transfer error" and "execution error" ?

Most likely not. It's similar to having a server in the Internet. If it does not respond to your requests anymore, then it is either a network connectivity issue or maybe the server itself has crashed or got powered down. But we can't really distinguish between these two cases in a reliable way.

In the case of FEL, we can try to make some guesses, but they will be never perfectly accurate.

closed 2026-09-08 by paulkocialkowski