linux-sunxi/sunxi-tools #101: fexc-bin -  product.: unknown type 34

vitalogy, 2017-04-10:
I know about the status of ```sunxi-script_extractor``` #69 #84, but i try to get a script.bin from A33 tablet and does not understand the error.

```
$ adb shell /data/local/tmp/sunxi-script_extractor > script_extractor.out
```

```
$ sunxi-fexc -v -I bin -O fex script_extractor.out script.fex
sunxi-fexc: from bin:script_extractor.out to fex:script.fex
fexc-bin: script_extractor.out: version: 1.2
fexc-bin: script_extractor.out: size: 131360 (79 sections), header value: 41512
E: fexc-bin: script_extractor.out: product.: unknown type 34
```

```
$ hexdump -C script_extractor.out | head
00000000  4f 00 00 00 28 a2 00 00  01 00 00 00 02 00 00 00  |O...(...........|
00000010  70 72 6f 64 75 63 74 00  00 00 00 00 00 00 00 00  |product.........|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  02 00 00 00 1a 03 00 00  70 6c 61 74 66 6f 72 6d  |........platform|
00000040  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000050  00 00 00 00 00 00 00 00  02 00 00 00 2e 03 00 00  |................|
00000060  74 61 72 67 65 74 00 00  00 00 00 00 00 00 00 00  |target..........|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000080  03 00 00 00 42 03 00 00  6b 65 79 5f 64 65 74 65  |....B...key_dete|
00000090  63 74 5f 65 6e 00 00 00  00 00 00 00 00 00 00 00  |ct_en...........|
```

```
$ sunxi-fexc -?
sunxi-fexc v1.4.2-48-gd9b1d7e
```

Other tries with ```sunxi-fel read 0x43000000 0x20000 script.bin ``` had also no success.

vitalogy, 2017-04-11:
You can find the file script_extractor.out [here](https://milaw.biz/files/ibowin_p130/script_extractor.out)

vitalogy, 2017-04-11:
I did a new try:

```
$ sunxi-fel version
AWUSBFEX soc=00001667(A33) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000

$ sunxi-fel write 0x2000 fes1.fex

$ sunxi-fel exe 0x2000

$ sunxi-fel read 0x43000000 0x10000 tab_script.bin

$ sunxi-fexc -v -I bin -O fex tab_script.bin script.fex
sunxi-fexc: from bin:tab_script.bin to fex:script.fex
fexc-bin: tab_script.bin: version: 1.2
fexc-bin: tab_script.bin: size: 65536 (79 sections), header value: 41512
E: fexc-bin: tab_script.bin: target.burn_key: invalid length 17 (assuming 1)
E: fexc-bin: tab_script.bin: power_sply.aldo2_6ol: invalid length 9 (assuming 1)
fexc-bin: Warning: Malformed entry key "sprite[gpio0"
E: fexc-bin: tab_script.bin: twi_para.twi_sda: invalid length 22 (assuming 6)
E: fexc-bin: tab_script.bin: clock.pll9: invalid length 65 (assuming 1)
E: fexc-bin: tab_script.bin: dram_para.dram_tpr3: invalid length 2049 (assuming 1)
E: fexc-bin: tab_script.bin: spi1.spi_cs0: invalid length 22 (assuming 6)
Segmentation fault

$ sunxi-fexc -?
sunxi-fexc v1.4.2-48-gd9b1d7e
```

ssvb, 2017-04-11:
The best way to resolve this is to look into the Allwinner kernel sources from their BSP. It does have a script.bin parser, so we can improve sunxi-tools to support these newer script.bin variants.

vitalogy, 2017-04-16:
Hmm ... where to get?
Is something like this useful: https://milaw.biz/wiki/android/ibowin#script_dump