linux-sunxi/sunxi-tools #104: Boot info not passed via Sunxi SPL

pascallj, 2017-07-07:
Currently in the process of getting mainline U-boot and kernel ready for the Topwise A721. When trying the FEL USBboot, I noticed that when uploading `boot.scr` using `sunxi-fel`, the `fel_scriptaddr` does not get set in the SPL header and `bootcmd_fel` fails.

The `sunxi-fel` tool fails with:
```
sunxi SPL version mismatch: found 0x02 > maximum supported 0x01
You need a more recent version of this (sunxi-tools) fel utility.
```

However when manually increasing `SPL_MAX_VERSION` in `fel.c` to `2` the tool works as it should and I get

```
Passing boot info via sunxi SPL: script address = 0x43100000, uEnv length = 0
```
and U-boot loads the script automatically.

What is the deal here? I assume increasing the `SPL_MAX_VERSION` is not an correct solution, but what is?

pascallj, 2017-07-07:
Maybe it's useful to mention that U-boot changed the `SPL_HEADER_VERSION` to `2` in [this commit](http://git.denx.de/?p=u-boot.git;a=commitdiff;h=7f0ef5a945a0a9cef773c2b05b157d2fef3ea7ad).

Seems like there were some noticeable changes in version 2 as well, so changing just the version number is probably not a solution. 
As I am just getting started with this project, I guess I don't have the skills to update the tool. Maybe someone can?

ssvb, 2017-07-07:
Thanks for reporting! This is indeed quite a nasty problem and we unfortunately dropped the ball earlier. Here are the links to older mailing list discussions:

https://groups.google.com/d/msg/linux-sunxi/XuwQF_cK2jA/BPw5F_q-AgAJ
https://lists.denx.de/pipermail/u-boot/2015-September/228727.html

Basically, the SPL version header bump in U-Boot should not have caused any disruptions for the existing users if we had done it right. But unfortunately right now all the sunxi-tools packages in various linux distributions will have to be updated because of this bug. I'll try to fix this ASAP and we will need to tag a new release too.

ssvb, 2017-07-07:
Yes, just changing SPL_HEADER_VERSION to 2 is not good enough, because this does not fix the root cause of the problem. And we can't play a catch-up game.

pascallj, 2017-07-07:
Well you basically identified my problem back in 2015:
> Then we can have the users trying to follow some
tutorial and expecting their boot.scr to be picked up by U-Boot. And
if boot.scr is just silently ignored, then they can have a lot of fun
troubleshooting or pestering people for support.

Only the error message of different versions made me look deeper into the issue.

You're right about the fact that a catch-up game is not the right way to play it. However the popularity of this whole project seems to have dropped quite a bit (even though I am just getting started, found this old Chinese tablet in a drawer), so how big of deal is it?

computermouth, 2017-07-17:
> However the popularity of this whole project seems to have dropped quite a bit (even though I am just getting started, found this old Chinese tablet in a drawer), so how big of deal is it?

Just for the sake of impressing that the project is still important, I'm the distribution maintainer for Next Thing Co. [ CHIP and CHIP Pro ]. I was popping over here because I wanted to see how our boards fared with mainline u-boot, and eventually ended up here with the same issue.

closed 2019-11-23 by pascallj