linux-sunxi/sunxi-tools #124: "You need a more recent version of this (sunxi-tools) fel utility"

mutluit, 2019-04-10:
I have the latest git version of sunxi-tools from this very place:

# ./sunxi-fel -h
sunxi-fel v1.4.2-67-g7fe6024

But it gives this error:

# ./sunxi-fel uboot u-boot-sunxi-with-spl.bin write 0x43100000 my.env
sunxi SPL version mismatch: found 0x03 > maximum supported 0x02
You need a more recent version of this (sunxi-tools) fel utility.

u-boot version is the latest 2019.04

A similar but older error situation is shown at http://linux-sunxi.org/FEL/USBBoot#Booting_the_whole_system_over_USB_.28U-Boot_.2B_kernel_.2B_initramfs.29 :

sunxi SPL version mismatch: found 0x02 > maximum supported 0x01.
You need a more recent version of this (sunxi-tools) fel utility.

mutluit, 2019-04-11:
Passing -v gives some more information, but it still fails:

./sunxi-fel -v uboot u-boot-sunxi-with-spl.bin write 0x43100000 my.env
Stack pointers: sp_irq=0x00002000, sp=0x00005E08
Reading the MMU translation table from 0x00020000
Disabling I-cache, MMU and branch prediction... done.
=> Executing the SPL... done.
Setting write-combine mapping for DRAM.
Setting cached mapping for BROM.
Writing back the MMU translation table.
Enabling I-cache, MMU and branch prediction... done.
Writing image "U-Boot 2019.04-00036-g1d63ec3fa4", 448324 bytes @ 0x4A000000.
sunxi SPL version mismatch: found 0x03 > maximum supported 0x02
You need a more recent version of this (sunxi-tools) fel utility.
Starting U-Boot (0x4A000000).

karlp, 2020-01-09:
This should have been fixed by the merge of #118

closed 2020-10-30 by wens