linux-sunxi/sunxi-tools #125: Orange Pi 3 cannot boot into fel-mode

detectingdragon77, 2019-04-15:
I am sorry but I cannot provide any logs because my Orange Pi 3 onboard emmc Android OS got corrupted when I tried to root it. My problem is that my Android OS installed on the EMMC of my Orange Pi 3 got corrupted. When I try to boot it into Android OS it only boots to the screen that says "Allwinner" in blue letters, then goes to a picture of mountains, then gets reboots and keeps doing the same thing over and over. It wont boot into an SD card OS, and I tried to connect the orange pi android os to my windows pc using usb otg, but it constantly loses connection when the reboot happens. I even tried flashing the "fel-sdboot.sunxi" file located (https://github.com/linux-sunxi/sunxi-tools/blob/master/bin/fel-sdboot.sunxi) on github. But when I inserted the sd card into the Orange Pi that is powered off, I turn it on and my computer wont recognize it. I also have a 64 bit linux pc on Debian. Please help, I have found nothing on the internet about "fel-mode on orange pi3". I am trying to install Android OS from the orangepi.org website: http://www.orangepi.org/downloadresources/OrangePi3/2019-01-23/orangepi3_0e7dae8bb1bf9678fda5754272.html If you need more information please reply and I well tell you anything I know.

nileshvora441, 2020-08-13:
@detectingdragon77 Did you find anything for FEL mode?

wens, 2020-10-30:
According to the schematic, the UBOOT pin is accessible from a pad on the board. You can use that to force the board into FEL.
For more information, please check the boot diagram in the user manual, and the linux-sunxi wiki. If you still have problems, please ask on IRC.

closed 2020-10-30 by wens