linux-sunxi/sunxi-tools #128: Force fastboot?

RenaKunisaki, 2019-06-17:
I'm trying to recover an A33 tablet (Logicom M BOT TAB 100) that's stuck with a broken recovery. It boots into TWRP which can't do anything (no touch screen or NAND support) whether holding buttons or not. The Android OS is still there, but I guess it's convinced it needs to boot into recovery mode on next boot and that flag is never getting cleared.

Because of that it's not even possible to enter fastboot mode to flash a working recovery, so it's essentially bricked. I am able to get into FEL mode and try to boot from SD card. (At least, if I put in SD card with bootloader on it, it boots to a blank screen instead of TWRP. I guess I haven't set it up properly, but it's doing something.)

Is there a way I can use FEL or u-boot to force the device to enter fastboot so that I can flash a working recovery image?

Nick768, 2021-05-30:
Sorry for hijacking this really old thread, but i have the same problem. I have an Allwinner A33 based Smartbook S9Q9 (buil.prop displayed Astar_h7 before i killed this device...) tablet and i tried to flash a twrp recovery image. Now it always boots into this broken recovery (same issue: no nand and no touchscreen support).

I've tested everything, but the only device i can manipulate is my sdcard. I can boot my tablet into fel mode (with a  sd card with fel-sdboot.sunxi flashed), but it doesn't respond to the sunxi-fel tool. Sunxi-fel always reports error -7 timeout (or something similar). Another problem is: there is no key combo for anything. You can't do anything, except boot it to recovery with the power button. Available buttons:
* Power on/off
* Back (it's a hardware key which can be pressed to boot to recovery)
* Volume +/- (no function while booting as far as i know)
* Reset button (in a small hole on the back of the device; (re-)starts the device instantly after releasing it)

There is and was no key combo to boot to fastboot. And the adb-command `adb reboot bootloader` doesn't work in twrp (i don't know why, but it just do a reboot to recovery).

Did you find a way to enter the fastboot mode again? Or does anybody else have a solution for this problem? I could provide more info like the downloaded build.prop, but i don't know if this is really necessary... And i have a backup (taken with dd) of the boot partition and the recovery partition. The main problem is, twrp doesn't see the nand devices.

Thanks in advise!

paulkocialkowski, 2026-10-19:
This is really about the stock firmware, not sunxi-tools.

closed 2026-10-19 by paulkocialkowski