linux-sunxi/sunxi-tools #130: Does 'sunxi-fel spl' supports allwinner t3 soc?

qianfan-Zhao, 2020-01-17:
Next is the log, the soc is T3.

```
$ sudo sunxi-fel -l 
[sudo] password for x: 
USB device 003:005   Allwinner R40     12c00002:84404700:78868500:381f014f
$ sudo sunxi-fel -v spl boot0_sdcard_sun8iw11p1.bin
[sudo] password for x: 
Stack pointers: sp_irq=0x00002000, sp=0x00005E08
MMU is not enabled by BROM
usb_bulk_send() ERROR -7: Operation timed out
```

qianfan-Zhao, 2020-01-18:
I had tested the SRAM region, seems only 0xBC00 ~ 0xC000 can't write.

Test SRAM by using this scripts:
```
#!/bin/sh

CHECK_REGION_NUM=100
CHECK_REGION_ADDRESS_LIMIT=48128 # 0xBC00

# Write random number(size is 1K) to address($1) and then check if the FEL
# work fine.
check_region_1k_writeable () {
	local address=$(echo "obase=16;$1" | bc)
	local check_n=0

	printf "Check region(1K) 0x%04X " $1
	bin1k=$(mktemp)

	while [ ${check_n} -le ${CHECK_REGION_NUM} ] ; do
		local pass=0

		dd if=/dev/urandom of=${bin1k} bs=1K count=1 &> /dev/null
		if sunxi-fel write "0x${address}" ${bin1k} > /dev/null; then
			if sunxi-fel version > /dev/null; then
				let check_n++
				echo -n "."
				pass=1
			fi
		fi

		if [ ${pass} = 0 ] ; then
			echo "FAIL"
			return 1
		fi
	done

	echo "PASS"
}

# Check the SRAM region between 0x0000 and 0xC000 if it is writeabale.
address=0

while [ ${address} -le ${CHECK_REGION_ADDRESS_LIMIT} ] ; do
	# Skip the SRAM region in SP_IRQ(0x1c00~0x2000) and SP(0x5c00~0x7000)
	if [ ${address} -ge 7168 -a ${address} -lt 8192 ] || [ ${address} -ge 23552 -a ${address} -lt 28672 ] ; then
		let address+=1024
		continue
	fi

	check_region_1k_writeable ${address}
	if [ $? -ne 0 ] ; then exit 1; fi
	let address+=1024
done

```

Next is the result:

```
sudo ./check_sram_writeable.sh
Check region(1K) 0x0000 .....................................................................................................PASS
Check region(1K) 0x0400 .....................................................................................................PASS
Check region(1K) 0x0800 .....................................................................................................PASS
Check region(1K) 0x0C00 .....................................................................................................PASS
Check region(1K) 0x1000 .....................................................................................................PASS
Check region(1K) 0x1400 .....................................................................................................PASS
Check region(1K) 0x1800 .....................................................................................................PASS
Check region(1K) 0x2000 .....................................................................................................PASS
Check region(1K) 0x2400 .....................................................................................................PASS
Check region(1K) 0x2800 .....................................................................................................PASS
[... 9 lines cut ...]
Check region(1K) 0x5000 .....................................................................................................PASS
Check region(1K) 0x5400 .....................................................................................................PASS
Check region(1K) 0x5800 .....................................................................................................PASS
Check region(1K) 0x7000 .....................................................................................................PASS
Check region(1K) 0x7400 .....................................................................................................PASS
Check region(1K) 0x7800 .....................................................................................................PASS
Check region(1K) 0x7C00 .....................................................................................................PASS
Check region(1K) 0x8000 .....................................................................................................PASS
Check region(1K) 0x8400 .....................................................................................................PASS
Check region(1K) 0x8800 .....................................................................................................PASS
Check region(1K) 0x8C00 .....................................................................................................PASS
Check region(1K) 0x9000 .....................................................................................................PASS
Check region(1K) 0x9400 .....................................................................................................PASS
Check region(1K) 0x9800 .....................................................................................................PASS
Check region(1K) 0x9C00 .....................................................................................................PASS
Check region(1K) 0xA000 .....................................................................................................PASS
Check region(1K) 0xA400 .....................................................................................................PASS
Check region(1K) 0xA800 .....................................................................................................PASS
Check region(1K) 0xAC00 .....................................................................................................PASS
Check region(1K) 0xB000 .....................................................................................................PASS
Check region(1K) 0xB400 .....................................................................................................PASS
Check region(1K) 0xB800 .....................................................................................................PASS
Check region(1K) 0xBC00 usb_bulk_send() ERROR -7: Operation timed out
FAIL
```

apritzel, 2020-01-19:
Hi,
I don't think the last KB of SRAM A4 not being writable is related to your issue. I don't know about the T3/R40 for sure, but some other SoCs need to flip some bits to enable CPU access to some SRAM regions. For the SPL we need at most 32KB anyway, so from 0x0000 - 0x7ffff.

So: are you sure that your boot0_sdcard_sun8iw11p1.bin file actually works for FEL booting? If this is an Allwinner boot0, I think it doesn't return to FEL, so the timeout is expected.
Please grab mainline U-Boot, and compile it for Bananapi_M2_Ultra_defconfig, for instance (any R40 board should do). Then use spl/sunxi-spl.bin as the SPL file. You should at least see some output on the serial, even if the DRAM init might not work (due to different DRAM configuration, for instance).

qianfan-Zhao, 2020-09-01:
@apritzel Sorry for the later response.

> are you sure that your boot0_sdcard_sun8iw11p1.bin file actually works for FEL booting? If this is an Allwinner boot0, I think it doesn't return to FEL, so the timeout is expected.

Yes, it's Allwinner boot0

> and compile it for Bananapi_M2_Ultra_defconfig

I had tried this defconfig and it can work fine. DRAM is OK, next is the log:

```
sudo sunxi-fel -v spl /tmp/sunxi-spl.bin
Stack pointers: sp_irq=0x00002000, sp=0x00005E08
MMU is not enabled by BROM
=> Executing the SPL... done.
```

```
U-Boot SPL 2020.10-rc3-13542-g23e333a5c0 (Sep 01 2020 - 11:00:43 +0800)
DRAM: 1024 MiB
Trying to boot from FEL
```

Thanks

closed 2020-09-01 by qianfan-Zhao