linux-sunxi/sunxi-tools #132: F1C100s support

mcejp, 2020-05-12:
Hello!

I see that there is code for F1C100s implemented in https://github.com/Icenowy/sunxi-tools/tree/f1c100s. Are there plans to merge it into master?
Is there a way I can help?

Thanks a lot.

Icenowy, 2020-11-23:
That branch has some features broken on non-ARM9 chips.

Maybe you can try to make machine codes for different architectures coexist.

giuliobenetti, 2022-01-30:
@mcejp @Icenowy I've just opened this PR:
https://github.com/linux-sunxi/sunxi-tools/pull/170 that rebase and adapt ARMv5TE to ARMv7+ so basically it resping @Icenowy work to be merged. Can you give a test so it would be easier to have merged?
Thanks in advance :-)

apritzel, 2022-01-31:
@Icenowy I actually tried what you suggested a few days ago: go with one compatible FEL thunk code for all of v5/v7/v8. That seemed to work for v7/v8, but reportedly not for the F1C100s.
Can you have a look, maybe? https://github.com/apritzel/sunxi-tools/commits/f1c100

giuliobenetti, 2022-01-31:
@apritzel @Icenowy i will take a look tomorrow since I have a debugger to see what is going on inside F1C100s as well as other SoCs :-)

apritzel, 2022-03-30:
F1C100s support was merged as #173. Closing.

closed 2022-03-30 by apritzel