linux-sunxi/sunxi-tools #135: H6 nand SPL too large?

akohlsmith, 2020-08-18:
I have an .img file that PhoneixUSB can flash to EMMC on a Rongpin H6B just fine, but if I'm using sunxi-fel to write `boot0_nand.fex` the tool complains that the SPL is too large (48kB). The `boot0_sdcard.fex` file I have is 32kB. I have several other .img files, and the NAND boot0 file is always 48kB, and the sdcard boot0 is 32kB.

I've noticed that both sunxi_tools and FELix are unable to properly use this SPL, but I'm still too new with the H6 to understand why this is an issue. It looks like the `sram_swap_buffers` specified in soc_info.c don't move the important areas "far enough over" to allow the remaining 16kB from the larger boot0_sdcard.fex file. I tried moving it over (changing buf2 in all the buffers from 0x2a400, 0x2a800 and 0x2bc00 to 0x2c400, 0x2c800 and 0x2dc00, and then also moving the thunk_addr over from 0x2a200 to 0x2c200 which *seemed* to work, but I'm not sure if this is the correct way to do this.

Icenowy, 2020-10-05:
The SRAM A1 of H6 is still only 32KiB. I don't think it will accept 48K SPL.

apritzel, 2020-12-23:
The H6 BROM actually supports loading up to 139 KiB of SPL from MMC. The H616 even goes higher. We actually need to deal with bigger SPLs for H616 support, and I am working on the proper support for that (have only some hacks for now).

Besides: I don't know why we need to copy the BROM *limitations*, on many SoCs we have the space for bigger SPL images. So even on H3 and A64 (where the BROM only support up to 32KB), we should allow larger SPLs.

apritzel, 2022-03-30:
Commit 2f59b574ba1047399 lifted the SPL size limit for the H6, it can now upload up to ~140KB. The U-Boot build system still knows a rather conservative SPL size limit, the only SoC going beyond 32KB is the H616. So any image created with mksunxiboot or mkimage is still limited to 32KB, even though this can be easily lifted there, at least for the H6, if needed. At the moment we keep this in place there to not endorse unnecessary code growth.