linux-sunxi/sunxi-tools #146: Speedup improvement for sunxi-fel by switching to high-speed USB device (?)

plaes, 2020-10-29:
```
07:42 < qdk0901> sunxi-fel can speed up simply by just re-enumerate as high speed usb device
07:44 < qdk0901> sunxi-fel writel 0x01c19040 0x29860
07:46 < qdk0901> writel((readl(MUSB_POWER) & ~MUSB_DEVCTL_SESSION) | MUSB_POWER_HSENAB, MUSB_POWER);
07:47 < qdk0901> by switch to usb high speed mode, fel transfer rate can up to 2800kB/s
```

deividAlfa, 2021-06-16:
Nice! For the F1C100s / F1C200s it also works, but different address:
sunxi-fel writel 0x01c13040 0x29960

qianfan-Zhao, 2021-06-30:
@deividAlfa R40 has the same address of R1C100s and the scripts can work in R40 platform.

But there's only a small improvement. I had checked the usb is enumed in usb high speed mode.

``` console
➜  fel-mass-storage git:(master) ✗ sudo ./start_r40_live.sh
[sudo] password for qianfan:
Sorry, try again.
[sudo] password for qianfan:
100% [================================================]  4232 kB,  349.9 kB/s
100% [================================================]    24 kB,  256.7 kB/s
100% [================================================]  5044 kB,  347.7 kB/s
Input this command in u-boot shell to boot linux kernel:
bootz 42000000 43300000 43000000
➜  fel-mass-storage git:(master) ✗ sudo sunxi-fel writel 0x01c13040 0x29860
➜  fel-mass-storage git:(master) ✗ sudo ./start_r40_live.sh
100% [================================================]  4232 kB,  546.0 kB/s
100% [================================================]    24 kB,  538.5 kB/s
100% [================================================]  5044 kB,  553.8 kB/s
Input this command in u-boot shell to boot linux kernel:
bootz 42000000 43300000 43000000
```

deividAlfa, 2021-07-01:
Well, that's 150% por original speed.
When you are uploading big binaries it helps a bit! 👍

qianfan-Zhao, 2021-07-01:
@deividAlfa Don't know how to speedup to 2.8M and which cpu are used? 

I had tested write emmc by using ums command and the speed can be up to 7M.  So the high speed usb device isn't the limiting. Maybe IBR or libusb is the speed bottleneck?

deividAlfa, 2021-07-01:
No idea. I guess because sunxi inits the most basic? So probably no dma?
It's hard to tell, allwinner devices have very poor documentation.

hno, 2021-07-01:
I would say a lot of the slowness is quite likely from the ping-pong protocol FEL is using, "write small block", "read response". To get full USB speed you need to stream the data, not use a request/response ping-pong protocol.

For large transfer I would recommend only transferring an U-Boot SPL with USB support using FEL, and then use USB device support in U-Boot to transfer the full u-boot and then the bulk data. This should give much better performance. But I have not attempted doing this so not sure how much work it is to prepare a such SPL, or if it even would fit in the available SRAM.

Note that the Allwinner tools also only transmits a small bootloader using the ROM FEL routines, the bulk transfers of the image are done using other USB transfer protocols, not the ROM FEL routines.

Quite similar to the method of using FEL for loading u-boot and then fastboot or ums in u-boot for transferring data to the eMMC, but targeting the DRAM instead of eMMC.

I am not entirely sure what you want to get done in sunxi-tools with this issue. Modifying the USB controller settings for specific devices is not entirely in the scope of sunxi-tools fel support, belongs more in device specific boot scripts or device documentation on the wiki.

hno, 2021-07-01:
Suitable wiki page for this: https://linux-sunxi.org/FEL/USBBoot

NikoDelarich, 2022-05-25:
Any ideas why this wouldn't work on my A64? The device re-enumerates in high-speed mode after writing 0x29860 to 0x1c19040 but the uboot command just hangs afterwards:
```
[ 2537.113504] usb 1-5.4.1: new full-speed USB device number 74 using xhci_hcd
[ 2537.218264] usb 1-5.4.1: New USB device found, idVendor=1f3a, idProduct=efe8
[ 2537.218272] usb 1-5.4.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
# sunxi-fel -v -p writel 0x01c19040 0x29860
[ 2540.935607] usb 1-5.4.1: new high-speed USB device number 75 using xhci_hcd
[ 2541.024022] usb 1-5.4.1: config 1 interface 0 altsetting 0 bulk endpoint 0x1 has invalid maxpacket 64
[ 2541.024031] usb 1-5.4.1: config 1 interface 0 altsetting 0 bulk endpoint 0x82 has invalid maxpacket 64
[ 2541.024038] usb 1-5.4.1: New USB device found, idVendor=1f3a, idProduct=efe8
[ 2541.024041] usb 1-5.4.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
# sunxi-fel readl 0x01c19040
0x00029970
# sunxi-fel -v -p uboot u-boot-sunxi-with-spl.bin
found DT name in SPL header: myDevice
Stack pointers: sp_irq=0x00012000, sp=0x00015E08
MMU is not enabled by BROM
usb_bulk_recv() ERROR -7: Operation timed out
```
I'm using the latest sunxi-tools ([commit 76089c82d0e1616aeb3b289790204dce98296477](https://github.com/linux-sunxi/sunxi-tools/commit/76089c82d0e1616aeb3b289790204dce98296477)).

jameshilliard, 2026-05-15:
I got this working with a few other optimizations on H616 in #234 by the way, managed to get over 20x speedup.

jameshilliard, 2026-05-17:
By the way I added support in #234 for a bunch more SoC's based on analysis of the [BROM dumps](https://github.com/smaeul/sunxi-blobs) but everything other than H616 is untested so far.