linux-sunxi/sunxi-tools #148: Support for V536

nandra, 2020-11-06:
Hi, I used sunxi-fel too to communicate with V536 based CPU but seems it's not fully supported as I'm getting:
```
 sudo ./sunxi-fel spiflash-info
Warning: no 'soc_sram_info' data for your SoC (id=1816)
SPI support not implemented yet for 0 ((null))!
marek@dnb:$ sudo ./sunxi-fel ver
Warning: no 'soc_sram_info' data for your SoC (id=1816)
AWUSBFEX soc=00001816(unknown) 00000001 ver=0001 44 08 scratchpad=00027e00 00000000 00000000
marek@dnb:$ sudo ./sunxi-fel sid
Warning: no 'soc_sram_info' data for your SoC (id=1816)
SID registers for your SoC (0x1816) are unknown or inaccessible.

```

So seems support is missing? Is there anything I can help with to add support? Thanks.

Icenowy, 2020-11-08:
Do you have the user manual of this SoC?

nandra, 2020-11-09:
@Icenowy yes it's available here: http://files.lindeni.org/lindenis-v536/Documents/SOC_brief_and_spec/V5x6_Professional_Camera_SoC_Datasheet_V1.2.pdf

Also more documentation here: http://wiki.lindeni.org/index.php/Lindenis_V536

nandra, 2020-11-09:
I've added some copy'n'paste stuff and also some data from datasheet:
```
diff --git a/soc_info.c b/soc_info.c
index 866e012..83670f5 100644
--- a/soc_info.c
+++ b/soc_info.c
@@ -125,6 +125,11 @@ sram_swap_buffers v831_sram_swap_buffers[] = {
        { .size = 0 }  /* End of the table */
 };
 
+sram_swap_buffers v536_sram_swap_buffers[] = {
+       { .buf1 = 0x21000, .buf2 = 0x38000, .size = 0x1000 },
+       { .size = 0 }  /* End of the table */
+};
+
 /* H616 situation is the same as V831 one, except it has 32 KiB of SRAM A1. */
 sram_swap_buffers h616_sram_swap_buffers[] = {
        { .buf1 = 0x21000, .buf2 = 0x28000, .size = 0x1000 },
@@ -291,6 +296,16 @@ soc_info_t soc_info_table[] = {
                .sid_offset   = 0x200,
                .rvbar_reg    = 0x09010040,
        },{
+               .soc_id       = 0x1816, /* Allwinner V536 */
+               .name         = "V536",
+               .spl_addr     = 0x20000,
+               .scratch_addr = 0x21000,
+               .thunk_addr   = 0x2A200, .thunk_size = 0x200,
+               .swap_buffers = v536_sram_swap_buffers,
+               .sid_base     = 0x03006000,
+               .sid_offset   = 0x200,
+               .rvbar_reg    = 0x09010040,
+       }, {
                .swap_buffers = NULL /* End of the table */
        }
 };
```

Seems there is no anymore warning about inknown machine and I can read SID. Issue is SDK from vendor came with 2014 u-boot and according: https://linux-sunxi.org/FEL/USBBoot for USB boot is necessary u-boot 2015 and higher. Is that right? Thanks.

Icenowy, 2020-11-09:
BSP will be never be able to do USBBoot. For USBBoot community ported U-Boot will be needed.

nandra, 2020-11-09:
@Icenowy you mean SDK u-boot? We would need mainline u-boot with V536 support? Thanks.

Icenowy, 2020-11-09:
Yes.

nandra, 2020-11-11:
@Icenowy I read more properly wiki and it mention also about booting legacy (version < 2015.04) using sunxi-fel tool (Legacy mainline U-Boot (v2015.04 and older versions))

I have u-boot 2014.07 with changes for CPU but following build steps it fails that it cannot find felconfig (tried also on mainline 2014.07 for Cubietruck and result is the same). Do you see a way to use this old u-boot for sunxi-fel? Thanks.

danielkucera, 2021-11-28:
Hi, I have an action camera Eken H9 which has chip V316 which seem so be very similar to V536, it identifies like this:
```
$ ./sunxi-fel ver
Warning: no 'soc_sram_info' data for your SoC (id=1816)
AWUSBFEX soc=00001816(unknown) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
```
There are also some resources available for this platform:
https://github.com/petesimon/allwinner-V316

I'd like to safely experiment with the device so I need to be able to dump/restore SPI flash via some means ideally FEL.
Is there any progress with the SPI flash support here?

danielkucera, 2021-11-28:
It seems I managed to make spi work, see #164 .

apritzel, 2022-03-30:
PR #164 was merged, so V536 is now officially supported.

closed 2022-03-30 by apritzel