linux-sunxi/sunxi-tools #153: sunxi-meminfo Error: unknown or unhandled Soc: 0x1680

heldiho53, 2020-12-28:
Have a mxq 4k pro 1gb/8gb allwinner h3 and this happens

apritzel, 2021-04-24:
sunxi-meminfo only supports older SoCs, not the H3 or anything later. The DRAM controller in those newer SoCs is different (actually changed twice since then, already), so the tool would require a serious update or rewrite anyway.
You can try register dumps from U-Boot (`md.l 0x1c62000`, and other addresses), but the register layout is different, so it requires more knowledge. What is the problem you are trying to solve? I think the mainline U-Boot DRAM driver for the H3 should work, at least I haven't heard of many problems with that.

apritzel, 2022-03-30:
Closing as a WONTFIX. Nobody will realistically add support for newer SoCs to this tool (but patches are always welcome!).

closed 2022-03-30 by apritzel