linux-sunxi/sunxi-tools #155: usb_bulk_send() ERROR -4: No such device

filipkotoucek, 2021-01-21:
Hello, 
I'm trying to boot new custom board with A64 via USB-OTG with this script (sunxi-fel v1.4.2-110-g7a6a222 compiled from source). New board uses LPDDR3, the previous board uses normal DDR3, so I'm using different SPL for each board.
```
sunxi-fel -v  $DEV \
        spl $SPL \
        write 0x44000 bl31.bin \
        write 0x4a000000 u-boot.bin \
        write 0x43100000 my.env \
        write 0x4FC00000 boot.scr \
        write 0x40080000 Image \
        write 0x4FE00000 live-image.cpio.gz.bin \
        write 0x4FA00000 sun50i-a64.dtb \
        reset64 0x44000
```
The script works on the previous boark OK. On the new one I'm getting `usb_bulk_send() ERROR -4: No such device (it may have been disconnected)` error.
Output on PC:
```
Image Name:   
Created:      Thu Jan 21 08:12:45 2021
Image Type:   ARM Linux Script (uncompressed)
Data Size:    193 Bytes = 0.19 KiB = 0.00 MiB
Load Address: 00000000
Entry Point:  00000000
Contents:
   Image 0: 185 Bytes = 0.18 KiB = 0.00 MiB
Image Name:   TestLinux initramfs root
Created:      Thu Jan 21 08:12:45 2021
Image Type:   ARM Linux RAMDisk Image (gzip compressed)
Data Size:    52587486 Bytes = 51354.97 KiB = 50.15 MiB
Load Address: 00000000
Entry Point:  00000000
Stack pointers: sp_irq=0x00012000, sp=0x00015E08
MMU is not enabled by BROM
=> Executing the SPL... done.
Passing boot info via sunxi SPL: script address = 0x43100000, uEnv length = 8
Passing boot info via sunxi SPL: script address = 0x4FC00000, uEnv length = 0
usb_bulk_send() ERROR -4: No such device (it may have been disconnected)
```

Otuptut from UART0:
```
U-Boot SPL 2018.11-00009-g4edd8c0427 (Dec 06 2018 - 01:23:58 +0000)
DRAM: 1024 MiB
Trying to boot from FEL

```
I'm not sure if it can be bug in `sunxi-fel` or electrical issue on the new board (voltage drop). Do you have any advice?

hno, 2021-01-21:
I would guess it is a board problem or SPL configuration error. Loading the
SPL to SRAM works, but then it crashes after the DRAM have been configured
where it should load u-boot into DRAM.

apritzel, 2021-04-24:
What SPL are you using? Latest mainline U-Boot supports 64-bit SPLs for FEL booting now, so no extra SPL needed. `sunxi-fel u-boot u-boot-sunxi-with-spl.bin` even works now.
One other problem I see is that sunxi-fel detects *two* magic scripts, so it gets confused and passes the wrong address, it looks like (Passing boot info ...). Can you boot without any scripts first, to rule out any issues with your binaries?
It looks like the initrd, wrapped in a U-Boot legacy image upsets it. Can you try to upload the pure binary version of it, then use `booti $kernel_addr_r $ramdisk_addr_r:0x3226bde $fdt_addr_r`?

paulkocialkowski, 2026-10-19:
Closing since there was no reply in years.

closed 2026-10-19 by paulkocialkowski