linux-sunxi/sunxi-tools #156: support for a63

ChalesYu, 2021-03-07:
Hi, I'm trying to boot A63 board using fel , but seems support is missing.
from linux-sunxi.org can know this soc is similar as h6 and a64. but now seems only BSP works.
Is there anything I can help with to add support?

guess it need a patch like this,  copy from h6

```
--- a/soc_info.c
+++ b/soc_info.c
@@ -307,6 +307,20 @@ soc_info_t soc_info_table[] = {
                /* Check L.NOP in the OpenRISC reset vector */
                .needs_smc_workaround_if_zero_word_at_addr = 0x100004,
                .watchdog     = &wd_h6_compat,
+       },{
+               .soc_id       = 0x1719, /* Allwinner A63 */
+               .name         = "A63",
+               .spl_addr     = 0x20000,
+               .scratch_addr = 0x21000,
+               .thunk_addr   = 0x42200, .thunk_size = 0x200,
+               .swap_buffers = h6_sram_swap_buffers,
+               .sram_size    = 144 * 1024,
+               .sid_base     = 0x03006000,
+               .sid_offset   = 0x200,
+               .rvbar_reg    = 0x09010040,
+               /* Check L.NOP in the OpenRISC reset vector */
+               .needs_smc_workaround_if_zero_word_at_addr = 0x100004,
+               .watchdog     = &wd_h6_compat,
        },{
                .soc_id       = 0x1817, /* Allwinner V831 */
                .name         = "V831",
```

apritzel, 2021-04-24:
What board is that? I have a Teclast M8 tablet, and from all I know the DRAM controller on the A63 is different from all of A64, H6 and H616, so it requires significant work to get support for that.
Initial PMIC support for the DRAM voltage is only something we need for the H616 so far, all other PMICs have sensible default voltages set on DCDC5 (or whatever is meant to drive the DRAM chips), so DRAM power works on reset.

ChalesYu, 2021-04-25:
@apritzel 

Happily found that board can flash Teclast M8 firmware , but AFAIK,  A63 only have one LPDDR3 variant, the main different should be the screen type, and other I/O port etc.

[Here](http://www.rervision.cn/%E5%85%A8%E5%BF%97%E7%B3%BB%E5%88%97a63-c-119_173/%E5%85%A8%E5%BF%97a63%E6%A0%B8%E5%BF%83%E6%9D%BF-p-154.html) is a A63 core board. Althrough some board has a eMCP for memory , but this type is still LPDDR3. So, shouldn't different on software driver(init DRAM) part. 

> Initial PMIC support for the DRAM voltage is only something we need for the H616 so far, all other PMICs have sensible default voltages set on DCDC5 (or whatever is meant to drive the DRAM chips), so DRAM power works on reset.

That helps a lot. I know why I can't find PMIC init code for A64+AXP803 in u-boot finally.

also, I found [a early libdram code](https://github.com/orangepi-xunlong/u-boot-orangepi/tree/a6f125d33abe232fab13abe414d15eba076ccae8/sunxi_spl/dram/sun50iw3p1/dram/lib-dram) (not binary),maybe it can helps.

closed 2021-08-25 by ChalesYu

ChalesYu, 2021-08-25:
Okay , seems this SoC is only for business use,  but vendor or official still want use BSP 3.10, I can't get any source code, and prepare to sell this board.

I even brought a H616 board (X96 mate),  and have tried to replace aw859a with ap6330, but failed.  stuck at probe axp305 . don't know how to fix it , now that h616 board has destroyed and lay in the trash.

So now game is over , hope others people have better luck play with these SoC .