linux-sunxi/sunxi-tools #157: error in make (fatal error: linux/libfdt.h: No such file or directory)

MahdadGhasemian, 2021-03-29:
$ make
cc -std=c99 -Wall -Wextra -Wno-unused-result -D_POSIX_C_SOURCE=200112L -D_BSD_SOURCE -D_DEFAULT_SOURCE -Iinclude/  `pkg-config --cflags libusb-1.0` `pkg-config --cflags zlib`  -o sunxi-fel \
	fel.c fit_image.c progress.c soc_info.c fel_lib.c fel-spiflash.c  `pkg-config --libs libusb-1.0` `pkg-config --libs zlib` -lfdt
fit_image.c:20:10: fatal error: linux/libfdt.h: No such file or directory
   20 | #include <linux/libfdt.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:141: sunxi-fel] Error 1

MahdadGhasemian, 2021-03-29:
Hi

I installed it by package and it work. :

`sudo apt install sunxi-tools`

zxc503, 2021-04-05:
sudo apt-get install libfdt-dev

giuliobenetti, 2022-01-30:
@MahdadGhasemian since this issue is resolved you should close it.

closed 2022-04-16 by MahdadGhasemian

artlog, 2022-11-05:
This information should be propagated here : https://linux-sunxi.org/Sunxi-tools#Building_from_Source