linux-sunxi/sunxi-tools #162: usb_bulk_send() ERROR -7: Operation timed out on Raspberry PI 3 B+

studiofuga, 2021-11-02:
When trying to boot from FEL mode on a perfectly working setup (cable and board) from a Raspberry PI 3B+ board, timeout happen:

```

pi@raspberrypi:~/sunxi-tools-1 $ ./sunxi-fel  ver
AWUSBFEX soc=00001667(A33) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000

pi@raspberrypi:~/sunxi-tools-1 $ ./sunxi-fel -v uboot u-boot-sunxi-with-spl.bin 
Stack pointers: sp_irq=0x00002000, sp=0x00005E08
MMU is not enabled by BROM
=> Executing the SPL... done.
Writing image "U-Boot 2018.05 for sunxi board", 549287 bytes @ 0x4A000000.
usb_bulk_send() ERROR -7: Operation timed out

```

According to [this thread](https://github.com/Thore-Krug/Flash-CHIP/issues/15#issuecomment-543924044), the issue may be related to the maximum transfert block size. Indeed changing from 512kib to 128kib fixes the issue.

```

pi@raspberrypi:~/sunxi-tools-1 $ ./sunxi-fel -v uboot u-boot-sunxi-with-spl.bin 
Stack pointers: sp_irq=0x00002000, sp=0x00005E08
MMU is not enabled by BROM
=> Executing the SPL... done.
Writing image "U-Boot 2018.05 for sunxi board", 549287 bytes @ 0x4A000000.
Starting U-Boot (0x4A000000).
pi@raspberrypi:~/sunxi-tools-1 $ 

```

tu95, 2022-04-25:
Though you use apt source to install sunxi-tools,  you will get error which tell you "usb_bulk_send() ERROR -7: Operation timed out". You should git clone sunxi-tools and build it from source code, you will solve this problem.

aslubsky, 2024-04-24:
The same story, increased the timeout, reduced the size from 512kb to 128kb, nothing helps, both the version with apt (v1.4.2) and the master branch (v1.4.2-178-gdf60a46-dirty) do not work. Any suggestions?