linux-sunxi/sunxi-tools #174: sunxi-fexc: Section name check too stringent

GMMan, 2022-03-07:
From what I can tell, the section name check is too stringent. I have a production script.bin that has section names such as `partitions/boot-res`, which sunxi-fexc rejects both `/` and `-`. Perhaps it's better to increase the allowed characters to printable non-space characters? I'm not even convinced the original fex compiler would reject spaces.

apritzel, 2022-03-08:
Should be addressed by PR #171, can you confirm that patch https://github.com/linux-sunxi/sunxi-tools/pull/171/commits/edec4d2f6c0 in there fixes this?

GMMan, 2022-03-18:
I think I manually made those changes to my copy shortly before filing the issue, so they should work, although I do still wonder if the check should be relaxed in general.

apritzel, 2022-03-30:
PR #171 has been merged, so this particular issue should be fixed now. I don't think we should become too relaxed here, as too loose naming tends to create all kind of ambiguities. If there is a real world case which we reject, but the original compiler passes on, this can be revisited.

closed 2022-03-30 by apritzel