linux-sunxi/sunxi-tools #181: Fit image introduction breaks building for windows

sgufa, 2022-08-01:
Hi,
I'm able to compile and run sunxi-fel on 32bit windows until commit #5541673db88f7a0182738e514665eb908738387a
After that, with the fit image support merge i cant compile anymore. Compiler has issues with fit_image.c and some libfdt references: 

fit_image.c: undefined reference to 
`fdt_getprop'
`fdt_get_property'
`fdt_path_offset'
`fdt_subnode_offset'
`fdt_get_property'
`fdt_path_offset'
`fdt_first_subnode'
`fdt_get_property'
`fdt_next_subnode'
`fdt_get_property'
`fdt_first_subnode'
`fdt_subnode_offset'
`fdt_get_property'

Any advice on how to solve this? I'm compiling with mingw-w32 under linux and with msys2 also under windows. same results.

apritzel, 2022-08-05:
You would need to install the headers and runtime library files (.so/.dll) for libfdt. I don't know from the top of my head how that works exactly in mingw, but I read that you would download the device-tree-compiler source (https://git.kernel.org/pub/scm/utils/dtc/dtc.git/snapshot/dtc-1.6.1.tar.gz) and build and install libfdt from there:
```
$ make libfdt
$ make install-lib install-includes
```
Not sure if your mingw installation comes with something where you can pull in compiled libraries more easily, but the above should work anyway.

sgufa, 2022-08-05:
Hello @apritzel 
already done. Also tried msys2 DTC library under windows. Same results: undefined references.
No one tried building for windows with FIT support?

apritzel, 2022-08-15:
So I managed to (cross-)compile a sunxi-fel.exe, from Slackware Linux. I don't know if there is something easier to install libraries, but I manually cross-compiled all required libraries, and then copied the .a and .h files into the SYSROOT. On my install this was under ```/usr/mingw-w64-v9.0.0_gcc11.2.0/x86_64-w64-mingw32/```. There are three directories in there: bin, lib, includes. I copied libz.a, libfdt.a and libusb-1.0.a into lib, and zconf.h, zlib.h, fdt.h, libfdt.h, libfdt_env.h and libusb-1.0/libusb.h into include. Then ```make CC=x86_64-w64-mingw32-gcc OS=Windows_NT``` created:
```sunxi-fel.exe: PE32+ executable (console) x86-64, for MS Windows```
I guess a 32-bit .exe would work similarly.
So did you use the static library? And made sure that libfdt.a ended up in the right directory?