linux-sunxi/sunxi-tools #182: usb_bulk_send() ERROR -7: Operation timed out on H616

nyatorch, 2022-10-09:
I bought a development board called [MangoPi MQ-Quad](https://linux-sunxi.org/MangoPi_MQ-Quad)
It has  H616 Soc and an empty SPI flash pad
[image]
[image]

I tried to compile mainline uboot for it, but there was something wrong when I tried to boot from FEL.

I used sun50i-h616-orangepi-zero2 as template so DT name is sun50i-h616-orangepi-zero2
```shell
$ sudo sunxi-tools/sunxi-fel ver
AWUSBFEX soc=00001823(H616) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000l
```
```shell
$ sudo sunxi-tools/sunxi-fel -v uboot u-boot-sunxi-with-spl.bin
found DT name in SPL header: sun50i-h616-orangepi-zero2
Stack pointers: sp_irq=0x00021400, sp=0x00053FD4
MMU is not enabled by BROM
=> Executing the SPL... done.
usb_bulk_send() ERROR -7: Operation timed out
```
and the serial output
```
U-Boot SPL 2022.10-00409-g2afa989fbe-dirty (Oct 07 2022 - 23:37:55 +0800)
pmic id is 0x4b
DRAM: 1024 MiB
Trying to boot from FEL
```
and then It stuck, no more output, sunxi-fel can't recognize fel device, I have to plug USB again.
To make sure it wasn't the compiled uboot that was the problem, I used someone else's compiled uboot.
```shell
$ sudo ../sunxi-tools/sunxi-fel -v uboot u-boot-sunxi-with-spl-armbian.bin
found DT name in SPL header: sun50i-h616-mangopi-mcore
Stack pointers: sp_irq=0x00021400, sp=0x00053FD4
MMU is not enabled by BROM
=> Executing the SPL... done.
usb_bulk_send() ERROR -7: Operation timed out
```
still stuck:
```
U-Boot SPL 2021.10-00409-armbian-dirty (Sep 29 2022 - 17:22:28 +0800)
pmic id is 0x4b
DRAM: 1024 MiB
Trying to boot from FEL
```

I also made some other attempts to try to solve the problem, including:
+ downgrade my binutils (binutils 2.39 has some rwx-segment warning, I have to modify Makefile to compile ATF and uboot)
+ connect my board to a USB2.0 Hub (my laptop all USB ports are USB3.0)
+ change USB_TIMEOUT in fel-lib.c from 10s to 60s
+ change AW_USB_MAX_BULK_SEND in fel-lib.c from 512KiB to 128KiB
+ replace another USB Cable

however, it still stuck at "Trying to boot from FEL".

#162 Mentioned with the installation source
but my sunxi-tools is git clone from this repository. I also tried AUR pakage: sunxi-tools-git, still not solved.

apritzel, 2022-10-09:
Hi, thanks for the report!
So just a quick check, to avoid talking about the wrong thing: did you add bl31.bin, so a Trusted Firmware-A binary, when compiling U-Boot? Since you mentioned another U-Boot binary, I doubt that this is the problem, but just checking.

We are investigating an apparent miscompilation or bug in the H616 DRAM driver. It only happens with GCC 11 or higher, and does not seem to affect booting from SD card, but only FEL (and apparently SPI) boot.
If that's easy enough for you to do, can you try compiling U-Boot with GCC 10 (or lower)? If that works, that's most likely the problem. We already spent some time starring at disassembled code from both GCC 10 and 11, without conclusive results.
I found one odd hack the other day: if you add a `udelay(0);` at the beginning of the `mctl_phy_read_calibration()` function in `arch/arm/mach-sunxi/dram_sun50i_h616.c`, it worked for me. If you could try that (with your current compiler), I'd be grateful to hear your results. But as long as we don't understand why this fixes the problem, I am bit reluctant to merge that fix.
Also if you could check that your current build works when booting from SD card, that would be a valuable data point.

As for the binutils problem: I am discussing this with some TF-A folks. We have some preliminary fix for the linker script, however that affects every BL31 build, and some platforms seem to be not compatible with this change. We will send a fix once this is solved.

Thanks, Andre

nyatorch, 2022-10-10:
> I found one odd hack the other day: if you add a udelay(0); at the beginning of the mctl_phy_read_calibration() function in arch/arm/mach-sunxi/dram_sun50i_h616.c, it worked for me.

It works for me, too. I'm using GCC 12.2.0 and binutils 2.39.3 on Arch Linux

(The PMIC on the board is actually an axp313, I just modified axp305.c)
```
U-Boot SPL 2022.10-00427-gf5717231ab-dirty (Oct 10 2022 - 21:45:28 +0800)
pmic id is 0x4b
DRAM: 1024 MiB
Trying to boot from FEL
NOTICE:  BL31: v2.7(debug):v2.7.0-519-gc19116dd6-dirty
NOTICE:  BL31: Built : 21:22:08, Oct 10 2022
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a0826e8, model: OrangePi Zero2
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
ERROR:   RSB: set run-time address: 0x10003
INFO:    Could not init RSB: -65539
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.

U-Boot 2022.10-00427-gf5717231ab-dirty (Oct 10 2022 - 21:45:28 +0800) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: OrangePi Zero2
DRAM:  1 GiB
Core:  49 devices, 17 uclasses, devicetree: separate
WDT:   Not starting watchdog@30090a0
MMC:   mmc@4020000: 0
Loading Environment from FAT... MMC: no card present
** Bad device specification mmc 0 **
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Hit any key to stop autoboot:  0 
MMC: no card present
=> 

```

nyatorch, 2022-10-11:
> It only happens with GCC 11 or higher, and does not seem to affect booting from SD card, but only FEL (and apparently SPI) boot.

The situation is the same as you said, booting a GCC 12.2.0 compiled U-Boot (without adding `udelay(0)`) from the SD card works fine, GCC 10.1.0 compiled U-Boot also can boot from FEL.

*I cloned the U-Boot repository again caused a different version number, but I also tested different methods on the same U-Boot version.*

```
U-Boot SPL 2022.10-00459-g73e741b8ee-dirty (Oct 11 2022 - 22:42:53 +0800)
pmic id is 0x4b
DRAM: 1024 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.7(debug):v2.7.0-519-gc19116dd6-dirty
NOTICE:  BL31: Built : 21:22:08, Oct 10 2022
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a082898, model: OrangePi Zero2
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
ERROR:   RSB: set run-time address: 0x10003
INFO:    Could not init RSB: -65539
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.

U-Boot 2022.10-00459-g73e741b8ee-dirty (Oct 11 2022 - 22:42:53 +0800) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: OrangePi Zero2
DRAM:  1 GiB
Core:  49 devices, 17 uclasses, devicetree: separate
WDT:   Not starting watchdog@30090a0
MMC:   mmc@4020000: 0
Loading Environment from FAT... Unable to use mmc 0:0...
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
** No partition table - mmc 0 **
Couldn't find partition mmc 0:1
```

Thank you for your help, otherwise it is unlikely that I would be able to understand this situation.

REevee0, 2023-01-27:
The solution fixed the problem for me. Thank you.