linux-sunxi/sunxi-tools #194: portable_endian.h not found?

silver2row, 2023-05-29:
Hello,

I seem to get an error when I use the command ` make fel ` from w/in the /sunxi-tools/ directory for the board I use.

Is there any known workaround that anyone knows of currently for this error?

I looked online and keep trying to research ideas about ` portable_endian.h ` but come up short. 

Seth

P.S. If you have solved this issue, please reply.

apritzel, 2023-05-30:
This is because `fel` is not a supported make target, so some **implicit** make rule will trigger, and compile the fel.c file without adding the local include/ directory, which contains portable_endian.h:

    $ make fel
    cc     fel.c   -o fel
    fel.c:19:10: fatal error: portable_endian.h: No such file or directory

The binary tool you want is actually called `sunxi-fel`, so use that as the make target, and it will add the correct command line arguments to the compiler call:

    $ make sunxi-fel
    Setting version information: v1.4.2-169-g6a2f6b5
    
    cc -std=c99 -Wall -Wextra -Wno-unused-result -D_POSIX_C_SOURCE=200112L -D_BSD_SOURCE -D_DEFAULT_SOURCE
    -Iinclude/  `pkg-config --cflags libusb-1.0` `pkg-config --cflags zlib`  -o sunxi-fel
    fel.c fit_image.c progress.c soc_info.c fel_lib.c fel-spiflash.c  `pkg-config --libs libusb-1.0` `pkg-config --libs zlib` -lfdt

Or just type `make` and it builds `sunxi-fel` as well, among the other tools.

silver2row, 2023-05-30:
Okay,

I use an Allwinner 64A as a target while the host is using `gcc (Debian 10.2.1-6) 10.2.1 20210110`.

Seth

P.S. I git clone the repo, use make, and everything works. But, then when trying to use `make fel` or `make sunxi-fel ` nothing really transpires.

silver2row, 2023-05-31:
Hello,

I went another route and got the board booted w/ u-boot ideas (sort of). Thank you for your time and effort in guiding me.

Seth

closed 2023-05-31 by silver2row

Andre-ARM, 2023-05-31:
@silver2row what do you mean with "trying to use ... make sunxi-fel nothing really transpires."? If you did `make` before, it builds sunxi-fel already. If you explicitly ask for `make sunxi-fel` again, it will not rebuild anything, because nothing has changed. So you don't need both a "naked" `make` and `make sunxi-fel`, just one of them.
If there is still a problem, please let us know.