linux-sunxi/sunxi-tools #198: Box is only detected during first seconds of power on

happeeshopper, 2023-08-01:
When connecting my Allwinner H6 box, it's only detected by Sunxi for the first 5 or so seconds after power on.

At power on I get valid responses:
```
sunxi-fel --list
USB device 001:014   Allwinner H6      82c00003:0c004708:0140cf59:3c772351

sunxi-fel version
AWUSBFEX soc=00001728(H6) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
```

Any attempt after that gives:

```
sunxi-fel --list
libusb_open() ERROR -1: Input/Output Error

sunxi-fel version
ERROR: Allwinner USB FEL device not found!
```
After these errors, the device appears to be still in FEL mode (light is red) and `lsusb` shows the device it's connected in FEL mode:

`Bus 001 Device 019: ID 1f3a:efe8 Allwinner Technology sunxi SoC OTG connector in FEL/flashing mode`.

I'm stumped, I wonder if there may be any help for this?

apritzel, 2023-08-01:
Does that happen if you power up, then just wait and issue the first `sunxi-fel ver` after >5 seconds, or is it actually the second sunxi-fel invocation that kills the communication?
Regardless, I think I saw this behaviour before (on some tablet, I believe), and was wondering if it's not actually the normal BROM FEL mode that you see initially, but already some custom bootloader code, just offering some FEL compatible protocol on the OTG port for a few seconds, then continuing booting (which maybe then fails)?
How do you trigger FEL mode? Sometimes this is not very reliable, and is sensitive to stray power. If in doubt, remove all external peripherals (UART, USB cable), trigger FEL mode, then re-connect the USB cable.

happeeshopper, 2023-08-01:
Ok I was mistaken, it isn't the power up or boot that's the trigger, it's the time after I let the tool off the FEL button (inside the audio jack). I presumed FEL mode stayed active once entered, but it seems it doesn't.

Even though the box looks like it's in FEL mode and `lsusb` shows it as being in FEL mode, apparently that's like a '2nd' FEL mode or something? If I press the hidden button again while it's in this mode, it's like the 1st mode is activated again and commands are sent/received again:

```
pc@amd:~
$ sunxi-fel --list
USB device 001:021   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:021   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:021   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:021   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:021   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:021   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
libusb_open() ERROR -1: Input/Output Error
pc@amd:~
$ sunxi-fel --list
libusb_open() ERROR -1: Input/Output Error
pc@amd:~
$ sunxi-fel --list
libusb_open() ERROR -1: Input/Output Error
pc@amd:~
$ sunxi-fel --list
USB device 001:024   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:024   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:024   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
libusb_open() ERROR -1: Input/Output Error
pc@amd:~
$ sunxi-fel --list
libusb_open() ERROR -1: Input/Output Error
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:026   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel version
ERROR: Allwinner USB FEL device not found!
pc@amd:~
$ sunxi-fel --list
libusb_open() ERROR -1: Input/Output Error
pc@amd:~
$ sunxi-fel --list
libusb_open() ERROR -1: Input/Output Error
pc@amd:~
$ sunxi-fel --list
USB device 001:029   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:029   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:029   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel --list
USB device 001:029   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~
$ sunxi-fel version
AWUSBFEX soc=00001728(H6) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
pc@amd:~
$ sunxi-fel version
ERROR: Allwinner USB FEL device not found!
```

The problem is, keeping the button pressed doesn't keep the valid FEL mode active indefinitely and it seems to cut off randomly.

The cable I'm using is new, works perfect with adb/fastboot/MTP etc.

Forgot to mention, I did add a udev rule, as recommended [here](https://github.com/zador-blood-stained/h3-fel-mass-storage):

`SUBSYSTEMS=="usb", ATTR{idVendor}=="1f3a", ATTR{idProduct}=="efe8", GROUP="sunxi-fel"`

I guess I'll try different commands when the mode is active to see if I can stop it from cutting off... tomorrow.

apritzel, 2023-08-01:
So the proper BootROM FEL mode is "eternal", it just stays there until the board resets(!) or some commands are send. So it could be that something is flaky, and the board resets for some reason, which then re-enters the BROM and thus FEL mode.
Or it isn't actually the BROM's FEL implementation, but something else, but that is speculation.

happeeshopper, 2023-08-04:
Unfortunately I'm not having any luck, the only commands that will work are the above ones, I tried booting a u-boot.bin but it won't have it. One time I got a "permission denied" when running the command without sudo so I've been trying it with sudo as well but it hasn't helped. 

`lsusb`  picks it up every time.

```
$ sudo sunxi-fel uboot u-boot-sunxi-with-spl-h6-noname.bin
usb_bulk_send() ERROR -7: Operation timed out
pc@amd:~/Desktop
$ sunxi-fel version
AWUSBFEX soc=00001728(H6) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000
pc@amd:~/Desktop
$ sudo sunxi-fel uboot u-boot-sunxi-with-spl-h6-noname.bin
usb_bulk_send() ERROR -7: Operation timed out
pc@amd:~/Desktop
$ sunxi-fel uboot u-boot-sunxi-with-spl-h6-noname.bin
ERROR: Allwinner USB FEL device not found!
pc@amd:~/Desktop
$ sunxi-fel uboot u-boot-sunxi-with-spl-h6-noname.bin
usb_bulk_send() ERROR -7: Operation timed out
pc@amd:~/Desktop
$ sunxi-fel --list
USB device 001:028   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~/Desktop
$ sunxi-fel --list
USB device 001:028   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~/Desktop
$ sunxi-fel --list
USB device 001:028   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~/Desktop
$ sunxi-fel --list
USB device 001:028   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~/Desktop
$ sunxi-fel --list
USB device 001:028   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~/Desktop
$ sudo sunxi-fel uboot u-boot-sunxi-with-spl-h6-noname.bin
usb_bulk_send() ERROR -7: Operation timed out
pc@amd:~/Desktop
$ sudo sunxi-fel uboot u-boot-sunxi-with-spl-h6-noname.bin
[sudo] password for pc: 
ERROR: Allwinner USB FEL device not found!
pc@amd:~/Desktop
$ sudo sunxi-fel uboot u-boot-sunxi-with-spl-h6-noname.bin
usb_bulk_send() ERROR -7: Operation timed out
pc@amd:~/Desktop
$ sudo sunxi-fel uboot u-boot-sunxi-with-spl-h6-noname.bin
usb_bulk_send() ERROR -7: Operation timed out
pc@amd:~/Desktop
$ sunxi-fel --list
pc@amd:~/Desktop
$ sunxi-fel --list
USB device 001:028   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~/Desktop
$ sunxi-fel --list
USB device 001:028   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~/Desktop
$ sunxi-fel --list
USB device 001:028   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~/Desktop
$ sunxi-fel --list
USB device 001:028   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~/Desktop
$ sunxi-fel --list
USB device 001:028   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~/Desktop
$ sudo sunxi-fel uboot u-boot-sunxi-with-spl-h6-noname.bin
usb_bulk_send() ERROR -7: Operation timed out
pc@amd:~/Desktop
$ sunxi-fel --list
USB device 001:029   Allwinner H6      82c00003:0c004708:0140cf59:3c772351
pc@amd:~/Desktop
$ sudo sunxi-fel uboot u-boot-sunxi-with-spl-h6-noname.bin
usb_bulk_send() ERROR -7: Operation timed out
```
I've been trying to catch it at the right time but it isn't having it.

Icenowy, 2025-11-12:
> I'm having this same issue on an Allwinner 0x1855 // AWUSBFEX soc==00001805(unknown). Using the trick described to hold the button down works for sending simple commands like version and list, but it's not effective for anything else. I'm also getting a warning about no 'soc_sram_info' data for your SoC (id-1855) when sending commands. The board is out of an SZ Naner touch screen smart music player.

0x1855 seems to be A133/R818.

BTW I suggest to open a dedicated issue for this instead of continue to use here.