linux-sunxi/sunxi-tools #200: sunxi-meminfo: kernel too old

ffpp2003, 2023-10-06:
Hi, I recently found this little tablet on one of my drawers and thought it could be a good device to have in working order (no android) for future projects. Since the wiki has no article about my specific device and shares no similarities other than the Soc, I decided to write the [corresponding article](https://linux-sunxi.org/X-View_Proton_Tab_2) for it.
Following the [New Device Howto](https://linux-sunxi.org/New_Device_howto) guide, I'm currently stuck at step 4. I was able to get the bootinfo.txt and script.fex files, but I cannot get the memory information. The sunxi-meminfo program seems to complain about my tablet having a "kernel too old". In its current (and I think stock) state, this tablet is running Android 4.0.4, using Linux version 3.0.8+. I saw no minimum Linux required version, so it doesn't make much sense.

paulkocialkowski, 2023-10-06:
Hi! I'm not seeing the error message in the code, do you have a more exact match for the error?

In general `sunxi-meminfo` will read the current values of registers using `/dev/mem`. Maybe support for `/dev/mem` was disabled in this kernel build? You could just check that the file exists.

In any case you might be able to extract the dram parameters from the fex that is used by the boot software (likely the one you got when retrieving the fex from fel mode).

ffpp2003, 2023-10-08:
This is the exact output I get:
```
fpp@FPP-PC:~/Documentos/proton2/sunxi-tools$ adb push sunxi-meminfo /data/cache/meminfo
* daemon not running; starting now at tcp:5037
* daemon started successfully
sunxi-meminfo: 1 file pushed. 6.0 MB/s (595868 bytes in 0.095s)
fpp@FPP-PC:~/Documentos/proton2/sunxi-tools$ adb shell
root@android:/ # cd data/cache/                                                
root@android:/data/cache # ls
meminfo
root@android:/data/cache # chmod 777 meminfo                                   
root@android:/data/cache # ls -la
-rwxrwxrwx root     root       595868 2023-09-25 08:47 meminfo
root@android:/data/cache # ./meminfo                                           
FATAL: kernel too old
Aborted 
134|root@android:/data/cache # 
```
The file /dev/mem exists, so, appearently the kernel feature is still enabled.

I've taken a look at the script.fex file i extracted, and found this:
```
[dram_para]
dram_baseaddr = 0x40000000
dram_clk = 384
dram_type = 3
dram_rank_num = 1
dram_chip_density = 2048
dram_io_width = 8
dram_bus_width = 16
dram_cas = 9
dram_zq = 0x7d
dram_odt_en = 0
dram_size = 512
dram_tpr0 = 0x42d899b7
dram_tpr1 = 0xa090
dram_tpr2 = 0x22a00
dram_tpr3 = 0x0
dram_tpr4 = 0x0
dram_tpr5 = 0x0
dram_emr1 = 0x0
dram_emr2 = 0x10
dram_emr3 = 0x0
```
They look to be ram parameters, but are those enough?

paulkocialkowski, 2023-10-08:
Oh I see! So this is not actually produced by the tool itself, it is produced by the C library complaining about the kernel version.
It means that you have built the tool with a toolchain that uses Linux header which are too new for the kernel you are running.

So you need to use a toolchain+libc compatible with Linux 3.0 here (e.g. try an older cross-compiler or build one for the purpose).

Otherwise the fex file should be enough, though it's good practice to check what is actually used by the controller at run-time (what meminfo does).

ffpp2003, 2023-10-08:
Ok, I might need a little bit of help here. It was quite a struggle to figure out the crosscompile stuff, now I gotta use an older one.

Running cat /proc/version on the tablet, I get this: `Linux version 3.0.8+ (gaoqingtao@brn-gaoqingtao) (gcc version 4.5.1 (Sourcery G++ Lite 2010.09-50) ) #57 PREEMPT Thu Jul 25 18:15:28 CST 2013`. So, I proceeded to download [this](https://sourcery.sw.siemens.com/public/gnu_toolchain/arm-none-eabi/arm-2011.03-42-arm-none-eabi-i686-pc-linux-gnu.tar.bz2). After downloading, I cleaned the source, ran `make tools`, and then proceeded to run `make target-tools CROSS_COMPILE=/home/fpp/Documentos/proton2/arm-none-eabi/bin/arm-none-eabi- PREFIX=/home/fpp/Documentos/proton2/arm-none-eabi/`, but I got this:
```
fpp@FPP-PC:~/Documentos/proton2/sunxi-tools$ make target-tools CROSS_COMPILE=/home/fpp/Documentos/proton2/arm-none-eabi/bin/arm-none-eabi- PREFIX=/home/fpp/Documentos/proton2/arm-none-eabi/
/home/fpp/Documentos/proton2/arm-none-eabi/bin/arm-none-eabi-gcc -std=c99 -Wall -Wextra -Wno-unused-result -D_POSIX_C_SOURCE=200112L -D_BSD_SOURCE -D_DEFAULT_SOURCE -Iinclude/ -static  -o sunxi-meminfo meminfo.c
meminfo.c:22:22: fatal error: sys/mman.h: No such file or directory
compilation terminated.
make: *** [Makefile:198: sunxi-meminfo] Error 1
```
I'm probably missing something with paths, but I wasnt able to figure it out by myself :/

paulkocialkowski, 2023-10-08:
Now this error indicates that the toolchain is missing some critical libc headers. That is because this build of the toolchain does not come with any libc (this is what the `none` part of the toolchain name indicates). Try again with another toolchain that has some libc with it.

ffpp2003, 2023-10-08:
Googling a little bit I found [this github](https://github.com/DiUS/arm-cs2011.03-42-toolchain) wich has a link for the arm-none-eabi compiler I'm using, but also has a arm-none-linux-gnueabi compiler. Using that solved my issue and now the tool prints the memory info correctly. Thanks for the help!

closed 2023-10-08 by ffpp2003