linux-sunxi/sunxi-tools #207: A133 support

parthitce, 2024-07-10:
I have a custom PCB based on A133 SoC with eMMC. Currently am trying to flash the device using sunxi-fel based on https://github.com/linux-sunxi/sunxi-tools/pull/192 and it resets the device. Also am new to the sunxi tools and still reading the documentation to get more understanding about what needs to be done.

1. Is it possible to flash spl-pub i.e boot0 and u-boot from vendor tree using sunxi-fel or it's only possible to flash mainline version?
2. https://github.com/linux-sunxi/sunxi-tools/pull/202#issuecomment-1824966240 @apritzel, did you managed to get this working after that point?

AFAIK the current mainline u-boot doesn't have support for A100 and A133 as well. Thanks in advance for the suggestions and comments.

apritzel, 2024-08-20:
1. sunxi-fel is mostly about *booting* a device, it only supports "flashing" to SPI NOR flash, not to SD card or eMMC. Its "uboot" command allows to upload both the SPL and U-Boot proper and boot that, but that only works with mainline versions of U-Boot. You can upload and execute boot0, but that won't get you very far, though this might initialise DRAM.
2. No, I didn't do any work on the A133, it's still on the far end of my TODO list, given the SoC's age and small significance.

lprot, 2025-03-18:
sunxi-fel even missed it's ID in the DB https://github.com/linux-sunxi/sunxi-tools/issues/219

parthitce, 2025-03-18:
> sunxi-fel even misses it's ID in the DB [#219](https://github.com/linux-sunxi/sunxi-tools/issues/219)

with https://github.com/apritzel/sunxi-tools.git and
```
sunxi-tools$ git diff
diff --git a/soc_info.c b/soc_info.c
index 2ecc23a..d4e8c73 100644
--- a/soc_info.c
+++ b/soc_info.c
@@ -629,6 +629,7 @@ soc_info_t soc_info_table[] = {
                .sid_base     = 0x03006000,
                .sid_offset   = 0x200,
                .needs_smc_workaround_if_zero_word_at_addr = 0x100004,
+               .rvbar_reg    = 0x08100040,
                .watchdog     = &wd_h6_compat,
        },{
                .swap_buffers = NULL, /* End of the table */
```
it's possible to load SPL and U-boot and also boot fine. U-boot with DRAM handling is used from https://github.com/BrokenR3C0RD/u-boot.git.

lprot, 2025-03-18:
> with https://github.com/apritzel/sunxi-tools.git and
>
> ```
> sunxi-tools$ git diff
> diff --git a/soc_info.c b/soc_info.c
> index 2ecc23a..d4e8c73 100644
> --- a/soc_info.c
> +++ b/soc_info.c
> @@ -629,6 +629,7 [...]

Can you merge ID into soc_info.c and tell me what to test? I have U-Boot of this device. If you need something from the stock fw it's here https://github.com/topjohnwu/Magisk/issues/8810

parthitce, 2025-03-18:
> Can you merge ID into soc_info.c and tell me what to test?
@apritzel is the original author for A133 changes. He can comment about the PR/merge.

lprot, 2025-03-18:
> > Can you merge ID into soc_info.c and tell me what to test?
> 
> [@apritzel](https://github.com/apritzel) is the original author for A133 changes. He can comment about the PR/merge.

I'm ready for whatever experiments as I have full stock image for PhoenixSuit and do not really need the device if we brick it. So just tell me what do to add full support of A133 :)

apritzel, 2025-03-19:
Please test https://github.com/linux-sunxi/sunxi-tools/pull/220, but please note that the BSP based firmware (U-Boot) will be not usable with sunxi-fel, as we rely on mainline U-Boot's ability to return to FEL after the DRAM init. Also sunxi-fel only knows how to parse the mainline U-Boot packages (legacy U-Boot and FIT image).
I was able to successfully boot the WIP mainline U-Boot port, either by @BrokenR3C0RD (https://github.com/BrokenR3C0RD/u-boot.git), or my version: https://github.com/apritzel/u-boot/commits/a133/, to be posted in the next days.

paulkocialkowski, 2026-02-03:
A133 support was merged and tested on multiple boards.

closed 2026-02-03 by paulkocialkowski