linux-sunxi/sunxi-tools #215: Writing to SPI flash is not working on F1C200s

Serhii-the-Dev, 2025-01-15:
1. A bootloader (properly patched with `mksunxi`) was written into SPI flash via:
```
.\sunxi-fel spiflash-write 0x0 .\simple_loader.bin
```
2. A device was power cycled
3. Device boots into FEL mode
4. Reading SPI flash via:
```
.\sunxi-fel.exe spiflash-read 0x0 25600 out.hex
```
The result (right) is clearly not matching the input (left):
[image]
Am I doing something wrong?

apritzel, 2025-01-15:
Does your device come with SPI *NAND* flash, by any chance? sunxi-tools only supports SPI *NOR* flash.

Serhii-the-Dev, 2025-01-15:
> Does your device come with SPI _NAND_ flash, by any chance? sunxi-tools only supports SPI _NOR_ flash.

Thank you, that makes sense. Alas I was not able to find specs of the memory chip I have, however according to [the Lctech Pi F1C200s spec on linux-sunxi.org](https://linux-sunxi.org/Lctech_Pi_F1C200s) the board has 128 MiB SPI-NAND.

Are there any chances to program it, or should I look for another dev boards (by any chance, could you please advise some)?

apritzel, 2025-01-16:
You should be able to write to the SPI NAND from Linux. The SPI NAND access should be generic, and the Allwinner SPI controller should work. However you won't be able to boot mainline U-Boot from it, as this does not support booting from SPI NAND at the moment. There are some old patches somewhere, but they would need to be updated, reviewed and tested. The SPI NAND flash on my LCtech Pi is apparently somehow broken, so I cannot help too much, at the moment. There are other boards using SPI NAND, though, and U-Boot (and eventually sunxi-fel) SPI-NAND support is board/SoC agnostic, so this will be supported sooner or later.

Serhii-the-Dev, 2025-01-17:
Thanks, I found one of those old patched `u-boot` instances and flashed it with `xfel`, it boots normally. There is also [a tutorial in the u-boot repository](https://github.com/ARM-software/u-boot/blob/master/board/sunxi/README.nand) which allows to program NAND flash using the sunxi-tools (basically by loading a specifically assembled image into RAM). I'm looking for a bare metal solution though, so currently struggling with a proper implementation of the `eGON` header for a very basic bootloader.
Thank you very much for pointing me in a right direction!

closed 2025-01-17 by Serhii-the-Dev

apritzel, 2025-01-17:
Please note that this "tutorial" you mention is about raw NAND flash, which is quite different from SPI NAND flash, software-wise, so it doesn't apply. It's only even been used and tested on older chips, and it looks like the F1C200s does not even support this parallel raw NAND flash anyway.
For a bare metal example, check out fel-sdboot.S and uart0-helloworld-sdboot.c in this repo. Adding an eGON header should be done with U-Boot's mkimage tool, or the old (and no longer maintained) mksunxiboot tool, if you really must.