linux-sunxi/sunxi-tools #222: H313 FEL support

carbofos, 2025-06-09:
I am trying to use FEL mode on TV BOX Wudung W01 which is similar to Tanix TX1 and Vontar QTV Q1. All are based on Allwinner H313, but it's detected as H616.

```
$ ./sunxi-fel ver
AWUSBFEX soc=00001823(H616) 00000001 ver=0001 44 08 scratchpad=00027e00 00000000 00000000

```

But it fails on uploading SPL:
```
$ ./sunxi-fel spl sunxi-spl.bin
usb_bulk_send() ERROR -7: Operation timed out

```

After fail ver also is not working until the board is not rebooted in FEL mode.
```
$ ./sunxi-fel ver              
usb_bulk_send() ERROR -7: Operation timed out

```

The same problem with u-boot feature:
```
$ ./sunxi-fel uboot u-boot-sunxi-with-spl.bin 
usb_bulk_send() ERROR -7: Operation timed out
```

Could you please fix FEL functions for H313 ?

PS: I did add Wudung W01 PCB images on Tanix TX1 page (both are similar):
https://linux-sunxi.org/Tanix_TX1

paulkocialkowski, 2025-06-09:
H313 and H616 are the same die (sun50iw9) so what sunxi-fel prints is correct.

Have you tried from a different host? The timeout issues may be related to the host USB stack (bootrom USB implementations tend to be a bit fragile) and not to an actual issue with sunxi-fel or the binaries.

ssvb, 2025-06-09:
> Have you tried from a different host? The timeout issues may be related to the host USB stack and not to an actual issue with sunxi-fel or the binaries (bootrom USB implementations tend to be a bit fragile).

The log says that executing the SPL was "done", so it was presumably successfully uploaded and executed. I would still do a test with the helloworld payload. If it works fine, then there are no problems on the sunxi-fel side.

carbofos, 2025-06-09:
One more thing:

When I use sunxi-fel v1.4.2 from Gentoo Portage i'm getting one scratchpad:

fel ver
AWUSBFEX soc=00001823(unknown) 00000001 ver=0001 44 08 scratchpad=00007e00 00000000 00000000

With master branch  sunxi-fel v1.4.2-195-g7540cb2  it's another:
AWUSBFEX soc=00001823(H616) 00000001 ver=0001 44 08 scratchpad=00027e00 00000000 00000000

> Have you tried from a different host?

I did try another host and got the same result.

> I would still do a test with the helloworld payload.

Could you explain what do you mean? What steps should be done?

apritzel, 2025-06-09:
So I just tried it on my Tanix-TX1: it all works as expected for me, with everything from its master branch:
- sunxi-tools commit 7540cb235
- uart0-helloworld-sdboot.sunxi as from the bin/ directory on github
- TF-A 06a5fe8e735
- U-Boot a854c12062c4
Built with some GCC 11.2 compiler, but please use a distro provided cross compiler. Don't use Linaro GCC builds anymore, they are out of fashion for years now. Cross toolchains from kernel.org or bootlin should work equally well, maybe just avoid the latest builds - GCC 15.0 for instance is quite new and less well tested.
Make sure you enter FEL mode correctly, it can be a bit touchy. What worked well for me is to boot into the internal firmware, type "su" to become root, keep the FEL button (in the AV socket) pressed, then type reboot and wait for "Restarting system with command 'shell'". Then try "sunxi-fel -v spl uart0-helloworld-sdboot.sunxi", that should return very quickly and print a hello message on the UART.
Hope that helps.

carbofos, 2025-06-11:
Just finished the test and UART output is:
```
Hello from Allwinner H616!
Returning back to FEL.
```

Does anybody have idea where is my problem now and how to make TV BOX to work with custom u-boot ?
I can try to recompile with crossdev GCC 14, but do not think it could change something.
I understand that I should close the ticket now but please let me know where I could continue such discussion.

> So I just tried it on my Tanix-TX1: it all works as expected for me, with everything from its master branch:

Did you try to boot compiled u-boot in FEL or on USB flash?

apritzel, 2025-06-15:
Do you get any output on the serial from the mainline SPL? If you get output from uart0-helloworld-sdboot.sunxi and from the vendor firmware, then you should see some messages from the SPL, for instance if the DRAM init fails, which would explain your symptoms.
For reference, try this (mainline) build of mine, for the TX1: https://paste.c-net.org/ThatcherSnowman
Also, can you please check the name of the Samsung chip on your device? I read something like KMD31000 on the (rather poor) picture in the Wiki, though that doesn't hit anything in Google. The TX1 uses a KMR31000 DRAM/eMMC chip, and the Wudung PCB read v2.0 (instead of v1.1 for the TX1), so there might be some differences in the DRAM setup. Also do you have any vendor update image that can be scanned for boot0 and its DRAM parameters?

carbofos, 2025-07-14:
> Do you get any output on the serial from the mainline SPL?

I did try to compile two repos compiled with aarch64-unknown-linux-gnu-gcc:
https://github.com/apritzel/u-boot/
https://source.denx.de/u-boot/u-boot.git

I do not have any problems with getting "Hello from Allwinner H616!" in my console but I'm getting timeout error with SPLs from both repos and the SPL provided in previous message link:
```
/usr/local/bin/sunxi-fel -v uboot u-boot-sunxi-with-spl.bin
found DT name in SPL header: allwinner/sun50i-h313-tanix-tx1
Stack pointers: sp_irq=0x00021400, sp=0x00053FFC
MMU is not enabled by BROM
=> Executing the SPL... done.
usb_bulk_send() ERROR -7: Operation timed out
``

My compiler:
```
aarch64-unknown-linux-gnu-gcc --version 
aarch64-unknown-linux-gnu-gcc (Gentoo 15.1.1_p20250705-r1 p2) 15.1.1 20250705
```

Exact chip name:
kmq310006a

I did not find firmware update but I have original dump ( I downloaded it using Android console from /dev/block/mmcblk0 . Full dump is huge:
https://uploadnow.io/f/G5DZcDm

Here are something what I found uisng binwalk.

system.dtb:
https://paste.c-net.org/MilnerWally

env:
```
arlyprintk=sunxi-uart,0x05000000initcall_debug=0console=ttyS0,115200nand_root=/dev/nand0p4mmc_root=/dev/mmcblk0p4init=/initloglevel=1selinux=0cma=64Mmac=wifi_mac=bt_mac=specialstr=keybox_list=hdcpkey,widevinesetargs_nand=setenv bootargs earlyprintk=${earlyprintk} initcall_debug=${initcall_debug} console=${console} loglevel=${loglevel} root=${nand_root} init=${init} partitions=${partitions} cma=${cma} snum=${snum} mac_addr=${mac} wifi_mac=${wifi_mac} bt_mac=${bt_mac} selinux=${selinux} specialstr=${specialstr} gpt=1setargs_mmc=setenv  bootargs earlyprintk=${earlyprintk} initcall_debug=${initcall_debug} console=${console} loglevel=${loglevel} root=${mmc_root} rootwait init=${init} partitions=${partitions} cma=${cma} snum=${snum} mac_addr=${mac} wifi_mac=${wifi_mac} bt_mac=${bt_mac} selinux=${selinux} specialstr=${specialstr} gpt=1boot_normal=sunxi_flash read 45000000 boot;bootm 45000000boot_recovery=sunxi_flash read 45000000 recovery;bootm 45000000boot_fastboot=fastbootrecovery_key_value_max=0x13recovery_key_value_min=0x10fastboot_key_value_max=0x8fastboot_key_value_min=0x2bootdelay=0bootcmd=run setargs_nand boot_normal
```

apritzel, 2025-07-16:
Yes, as you figured Allwinner BSP firmware never tries to read anything from USB. Indeed the Tanix TX1 and this box are a bit of a challenge, since they are one of the very few without an SD slot, which normally makes AW based boxes quite hackable. Do you have root access on the vendor firmware? I guess you could then try to download the U-Boot image to the device (via a webserver or file sharing or via an SSH server), then try to get a console and call dd from there.

The other (more involved) solution would be to compile a 32-bit U-Boot (or at least SPL), then load and execute this via FEL, then upload the actual 64-bit image to DRAM and use the booted 32-bit U-Boot to write this to the eMMC boot partition. This might work without leaving the non-secure world that you are trapped in. I did compile 32-bit U-Boot builds for 64-bit SoCs in the past, but I haven't tried lately to see what kind of hacks are still needed to make this work.
This scheme is based on the fact that you don't need secure or 64-bit mode to access the eMMC, this is all readily available from non-secure AArch32. Theoretically you can program all needed MMIO registers via FEL, but it's extremely tedious, of course. I heard of people working on direct MMC access via sunxi-fel (similar to how we can write to NOR flash this way), but I don't think this went anywhere. But there should be no technical hurdles preventing this.

apritzel, 2025-07-20:
So I think I found a way, based on the 32-bit build idea above. Please try the following:
- Build a normal firmware version of mainline U-Boot (plus TF-A) for the Tanix TX1, but set ``CONFIG_SPL_IMAGE_TYPE_SUNXI_TOC0=y`` in menuconfig or the defconfig file. Copy the resulting ``u-boot-sunxi-with-spl.bin`` somewhere safe.
- Apply this patch (on top of mainline): https://gist.github.com/apritzel/573e1607b255f23ee025837379d8c706
- Do a ``$ make clean``; then re-configure for the Tanix TX1 again (make sure CONFIG_SPL_IMAGE_TYPE_SUNXI_TOC0 is not set)
- Now switch your cross-compiler to an arm32 bit compiler, and compile U-Boot again.
- Verify with ``$ file spl/u-boot-spl u-boot`` that both are 32-bit binaries.
- Bring your box into FEL mode (turn on with hidden FEL button in headphone jack pushed)
- Boot the box with the 32-bit build, and upload the real 64-bit build into DRAM:

``$ sunxi-fel -v -p spl spl/sunxi-spl.bin write 0x4a000000 u-boot-dtb.bin write 0x50000000 /path/to/saved/64-bit/u-boot-sunxi-with-spl.bin exec 0x4a000000``

- Write the uploaded 64-bit binary to the eMMC boot partition:

    => mmc dev 1
    => mmc bootbus 1 1 0 0
    => mmc partconf 1 1 1 1
    => mmc write 0x50000000 0 0x7f0

- reset the board: ``=> reset``, now it should boot into mainline U-Boot. Load a kernel from a USB drive and enjoy.
- To get back to the vendor BSP firmware, boot into mainline U-Boot, then type: ``=> mmc partconf 1 1 0 0``, then reset.
- To re-activate the mainline firmware, boot via FEL into the 32-bit firmware (no need to upload the 64-bit build again), then just re-activate the eMMC boot partition: ``=> mmc partconf 1 1 1 0``, then reset.

Hope that works, I tested this on a secure-boot A133 tablet, where FEL has the exact same problem as your box.

paulkocialkowski, 2026-02-03:
Maybe make sure u-boot has selected the correct mmc device with `mmc info`.
Perhaps you need to use `mmc dev 2` instead of `mmc dev 1`.

I've also renamed the thread since this is not an H313-issue but really trying to boot via FEL on a TOC0 device.
Also see the wiki's [TOC0 page](https://linux-sunxi.org/TOC0#Implement_and_test_FEL_boot_for_TOC0_bootloaders.) (this feature is not yet supported).

paulkocialkowski, 2026-10-19:
PR #255 should allow booting on "secure" H616/A133 devices, so the last remaining piece to solve this is to handle the TOC0 image format in sunxi-tools.

But building an eGON image instead of TOC0 should already allow starting u-boot via fel on such a "secure" device!